ArduPilot/ardupilot #5569: EZ-UHF PPM Not Detected (Pixracer)

brizey02, 2017-01-16:
#### Issue details
EZ-UHF is not detected on the RC In, Confirmed Taranis, EZ-UHF is working and setup. 

#### Version
_What version was the issue encountered with_

#### Platform
[  ] All
[  ] AntennaTracker
[ X ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
Quad

#### Hardware type
 pixracer,

OXINARF, 2017-01-20:
Our dev team member @lvale tested this and it worked for him. According to the post in our forum it was found to be a bad clone.

closed 2017-01-20 by OXINARF

brizey02, 2017-01-20:
Thanks, Sorry I just found out it was a clone a few hours ago!

lvale, 2017-01-21:
Responding to your question on RCG I'm using these http://www.immersionrc.com/fpv-products/ezuhf-4ch-receiver/
with these modules
http://www.immersionrc.com/fpv-products/ezuhf-jr-module/
These are configured to only 8 PPM channels

brizey02, 2017-01-21:
Thanks, hope to get a real pixracer soon and get up and going on my ez-uhf!