ArduPilot/ardupilot #5570: Tailsitter VTOLs orphaned? It should be Arduflight.

wazoo22, 2017-01-16:
After 60 years building and flying all kinds of airplanes, big and little, ---here I am.
I'm a recent convert to the arducopter/arduplane package, trying to find software/firmware to fly tailsitter VTOLs. The problem that crops up right away is the division between arducopter and arduplane. 
Right now the firmware for VTOLs is quite limited in both. What there is is aimed at a multicopter with a plane grafted into/onto it. There is certainly some interest in this version of the VTOL concept, but seems that some  modifications would make the firmware usable for tailsitters as well. Tailsitters are currently either multi-engine tractor concepts or twins that are coaxial contra-rotating. Neither are covered by this firmware at present, but could be, thereby opening up a fertile field of design that has been sorely neglected, and has great utility and simplicity.  
I personally would support such firmware enthusiastically, with prototyping. I have several already-flying examples, using the KK 2.1.5 board, that really could use some better hardware like the Pixhawk or Pixfalcon, if firmware were available. The coaxial firmware that's in the testing stages in Copter has great merit but lacks the transition and functionality to function as a plane. Can we get these efforts together somehow?
_Please describe the problem, or desired feature_

#### Version
_What version was the issue encountered with_

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_

#### Logs
_Please provide a link to any relevant logs that show the issue_

tridge, 2017-01-16:
hi @wazoo22 we don't yet support tailsitters, but I expect we will in the next few months. If you would like to be a tester for when we do add support that would be appreciated.
The things we need to do to make this work are:

- support the frame type in hover using the copter code, make sure it flies well as a copter
- add that as a supported frame type to the existing quadplane code. That is very easy to do.
- add support for changing the 'level' reference when transitioning

@lthall and I have been discussing supporting tailsitters for a while, and it really is just a matter of a bit of time to implement it.

wazoo22, 2017-01-17:
I'll be happy to act as a tester. Looking forward to it.
That said, I've made it plain that I'm a novice with Ardu(whatever), so 
please bear that in mind. I'm also not a coder/programmer, so I'll 
surely be the perfect partner to see if the firmware is comprehensible 
to the ordinary mortal. I will say that anyone who can climb the 
learning curve of OpenAeroVTOL on the KK2.1.5 is a good climber!
I'm a serious builder and a quick study (for an ancient retread from the 
mists of time). If you want me to help, and feel I CAN help, I'd suggest 
that you round up someone with lots of patience who will answer my 
newbie questions--or face the music yourself! ;-)
I also like to write, and will be glad to help with the documentation.

I strongly feel that another incestuous marriage between a copter frame, 
be it quad/tri/etc. and an airplane will not really accomplish much. 
That already exists, in the quad/aircraft form, and needs only to be 
polished vis. transitions, and the opening up of the full range of 
parameters to the aircraft once it transitions to a wing-supported mode. 
In my view, the area to dive into right now to get the drop on the trend 
is tailsitters that use their main thrust system as a lift system long 
enough to get into the plane mode, or back on the ground safely. No tilt 
anything, except the entire airframe-too complex, and therefore too 
failure-prone. My two current flying prototypes do just that. Wherever I 
post photos, whenever I fly in public, I'm swamped with emails and 
questions from people who also would like the ease of launch and 
retrieval that is so important to a disabled person. A lot of people 
just think they're way cool, and that's enough to fire them up.
For me, the ideal would be the coaxial copter and maybe the mono-copter 
as a supported frame types, and a marriage of Arducopter and Arduplane. 
Consummated when you flip that transition switch, and your aircraft 
undergoes a transformation. I call my second proto "Changeling".
Thanks for the note.
Jim

tridge, 2017-06-16:
we support tailsitters in 3.8 beta
See http://discuss.ardupilot.org/t/dual-motor-tailsitters/15302
thanks for your patience!

closed 2017-06-16 by tridge