ArduPilot/ardupilot #5575: Quadplane Request: Simple Mode

iskess, 2017-01-17:
Using WV_GAIN, the plane's heading is controlled by the wind direction. This can make it more difficult for users who don't have much Copter experience to maneuver the Quadplane in QHover to a landing. 
Additionally, Quadplanes typically have poor yaw control, which makes using WV_GAIN helpful by allowing the yaw to drift with the wind direction. 
Allowing the rudder to operate in QHover and QLoiter modes will help to make yaw control better. (A shameless plug for my issue #4459)

magicrub, 2017-01-17:
I thought it worked in all modes, because it's basically a copter when in any quadplane mode? Have you tried a very low WV_GAIN? Anyhow, this seems like a duplicate of https://github.com/ArduPilot/ardupilot/issues/4459 so lets close this and discuss on the other issue

iskess, 2017-01-17:
Thank you for your quick response. 
I cannot find a parameter to control Simple mode among the Q_'s. I don't believe it has been implemented for Quadplane. This is a different issue from #4459 which has to do with control surface movement, not direct user stick input. I think they need to be tracked separately.

magicrub, 2017-01-17:
OK, let's track separately