ArduPilot/ardupilot #5582: compass calibration issues

csurf, 2017-01-19:
#### Issue details
I'm having problems doing mag calibration for APM plane w/ latest versions of arduplane & mission planner when using either live calibration or onboard calibration. 

- Live calibration sometimes fails with a .NET 'array index out of bounds' errors. The 'new offsets' message box still appears, and the new offsets are apparently successfully saved, but the procedure seems to bomb  & error out early due to the out-of-bounds error.

- onboard calibration completely hangs after 1-2 seconds on 3.71; the 1-second notification 'beeps' simply stop, and board becomes unresponsive. Disconnecting/reconnecting the COM port doesn't help. The board must be power cycled. I loaded arduplane version 3.60, and on-board mag calibration works fine with that version.

#### Version/Hardware
- Mission Planner 1.3.43 build 1.1.6202.10835
- Arduplane 3.71/3.60
- pixhawk w/ external mag on GPS module

WickedShell, 2017-01-29:
The live calibaration problem is a Mission Planner issue, and should be posted on the Mission Planner repo: https://github.com/ArduPilot/MissionPlanner

Regarding the onboard side, I've been unable to reproduce this both with plane 3.7.0 and master. (Master isn't a fair comparison as a large number of compass cal fix ups happened after the 3.7 release, but none of them should have been a crash fix, they were all reporting errors that were addressed. Others are reporting the same problem as well though: http://discuss.ardupilot.org/t/compass-onboard-calibration-with-pixracer-plane-3-7-1/13981 so we need to keep trying to chase it down.

meee1, 2017-01-29:
please provide the exact error message missionplanner throws. screenshot is fine

pjwlk, 2017-02-25:
I have exactly the same issue. (onboard calibration completely hangs after 1-2 seconds on 3.71; the 1-second notification 'beeps' simply stop, and board becomes unresponsive.)

The last attempt I changed the baud rate to 57600 received the error message in the attached image instead of the crash.

[image]
[image]

govsux, 2017-03-06:
I am having a similar problem. Cannot do accel calibrations. Timeout keeps happening it never lets you move forward. ArduPlane V3.7.1 (22b5c415)    Tried two different GCS with separate apps. APM Planner and Qground control iOS. Seemed related ?? so I posted it here...

pjwlk, 2017-03-12:
I finally found some relief on my issues.  I'm not completely through it yet because I'm short a air speed sensor cable but I'm confident I can now get the autopilot working.

I decided to try QGroundControl.  Grasping at straws, I changed the firmware on my Chinese Pixhawk to the  choice of "Ardupilot Flight Stack" (currently shows as Standard AERT Plane firmware version 1.5.5) and fired up the setup routine, which BTW is far far easier to do than using the Mission Planner app.  I now have everything calibrated except my airspeed sensor of course.  I can't find a reference to the exact firmware at this time for some reason.  The pixhawk now appears to work with Mission Planner - no toilet bowling on the compass, the GPS works correctly and the accelerometers are also setup correctly.  

The initial firmware install from Mission Planner would crash when trying to set up the compasses using the internal routine. The MP routine would never finish.  Hope it helps you.

WickedShell, 2017-11-05:
This appears to be a support issue.

closed 2017-11-05 by WickedShell