ArduPilot/ardupilot #5583: build issue

Tom-Shen, 2017-01-19:
#### Issue details
I follow the tutorials to build my first example program:
http://ardupilot.org/dev/docs/learning-ardupilot-the-example-sketches.html

Command is below:
./waf build --target examples/INS_generic --upload

Error code is below:
Waf: Entering directory `/Users/jerrys/Ardupilot-Tomshen-Git/ardupilot/build/px4-v2'
Excluding oreoled.bin
[ 5/13] Processing modules/mavlink/message_definitions/v1.0/ardupilotmega.xml
Traceback (most recent call last):
  File "/Users/jerrys/Ardupilot-Tomshen-Git/ardupilot/modules/mavlink/pymavlink/tools/mavgen.py", line 16, in <module>
    from pymavlink.generator import mavgen
  File "/Users/jerrys/Ardupilot-Tomshen-Git/ardupilot/modules/mavlink/pymavlink/generator/mavgen.py", line 12, in <module>
    from future import standard_library
**ImportError: No module named future**

mavgen returned 1 error code
Waf: Leaving directory `/Users/jerrys/Ardupilot-Tomshen-Git/ardupilot/build/px4-v2'
Build failed
 -> task in 'mavlink' failed (exit status 1): 
	{task 4332864768: mavgen ardupilotmega.xml -> }
''

#### Version
_What version was the issue encountered with_

#### Platform
MacOS 10.12
[  ] All
[  ] AntennaTracker
[ X ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_
pixhawk

#### Logs
_Please provide a link to any relevant logs that show the issue_

roger-zhao, 2017-01-19:
Hi, Tom, just install future module which is package of Python

Tom-Shen, 2017-01-19:
Hello, Roger

Thank you very much, I have installed the future module, then I got below error for the same command:

localhost:ardupilot jerrys$ ./waf build --target examples/INS_generic --upload
Waf: Entering directory `/Users/jerrys/Ardupilot-Tomshen-Git/ardupilot/build/px4-v2'
Excluding oreoled.bin
**[12/13] CMake Build px4 msg_gen
make: *** No rule to make target `msg_gen'.  Stop.**

Waf: Leaving directory `/Users/jerrys/Ardupilot-Tomshen-Git/ardupilot/build/px4-v2'
Build failed
 -> task in 'px4_msg_gen' failed (exit status 2): 
	{task 4578913728: cmake_build_task  -> }
['/usr/local/bin/cmake', '--build', '/Users/jerrys/Ardupilot-Tomshen-Git/ardupilot/build/px4-v2/modules/PX4Firmware', '--target', 'msg_gen']
localhost:ardupilot jerrys$

OXINARF, 2017-01-19:
@Tom-Shen Try a `make px4-clean`, followed by a `./waf clean` and then try to build it again.

This is a development support question and those should be posted in the forum (http://discuss.ardupilot.org) or asked about in Gitter (http://gitter.im/ArduPilot/ardupilot). For this reason, I'm closing it.

closed 2017-01-19 by OXINARF