ArduPilot/ardupilot #5585: Non-Persistent control params since RC Channel Split

jberaud, 2017-01-19:
#### Issue details
From commit https://github.com/ArduPilot/ardupilot/commit/e85df47b065d5a426a57404bf45c3ec438a2e825 ATC_RAT_* and some other params have become non persistent, i.e writing them has no effect after startup.
Then after commit https://github.com/ArduPilot/ardupilot/commit/d430cd62a6915cb5bc0e379f9ffe63106ac9bfba params are not loaded from default file when they are already present. The effect is that they are not even loaded from default param file but remain at their default value (hardcoded).

#### Version
master

#### Platform
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover

#### Airframe type
Bebop

#### Hardware type
Bebop

gmorph, 2017-01-19:
Do we know if this is this only occurring on the Bebop?
Thanks, Grant.

tridge, 2017-01-20:
i've reproduced the problem with changing ATC_RAT_RLL_P in SITL on master. I'm working on finding the issue and fixing
Thanks for letting us know!

tridge, 2017-01-20:
a PR is pending to fix this in #5586

tridge, 2017-01-20:
this is fixed in master now. Very good catch Julien!

closed 2017-01-20 by tridge