ArduPilot/ardupilot #5597: No IO Thread Heartbeat

tridge, 2017-01-21:
We occasionally see "No IO Thread Heartbeat"
I just reproduced this on Disco and tracked down the cause in that case to being that the _log_directory default had changed, so that DataFlash_File::Init() returned early before the IO thread was registered.
I think the error message should either not be disaplayed, or should be different if we have never called register_io_callback()

ferdicakmak, 2017-05-14:
I have seen this error with ArduPlane V3.8.0beta5 (which was not present with v3.7.1) together with GCS message "Bad Logging". Although I formatted SD card, the problem is still present occasionally. 

Any suggestions?

kkouer, 2017-05-14:
I got this error too
In the mission planner i found where this message show, in the code i can see sensor health bad but don't  know which sensor.

Kkouer iPhone 6

EShamaev, 2017-05-14:
Today during flight test of latest master I encountered "No IO Thread Heartbeat" 4 times out of 6 flights. Each time after that the SD card logging stopped working with "Bad Logging" message and one time after that PWM outputs stopped working. This was on PX2.

robustini, 2017-05-20:
Same problem here in two different VTOL, "Bad Logging"...

ferdicakmak, 2017-06-07:
I have tried 3 different brand new sd cards on Pixhawk with Arduplane V3.8.0beta5 (QuadPlane config), and "No IO Thread Heartbeat problem still exists. I have also formatted sd cards several times., but nothing changed. 
I wonder if it is a firmware issue, which was not present with Arduplane V3.7.1 and earlier versions. 
Any suggestion is welcome.

ferdicakmak, 2017-06-07:
I have reduced LOG_FILE_BUFSIZE to 4 and "No IO Thread Heartbeat" problem seems to be solved, at the expense of lower logging rate.

While Pixhawk connected to PC over USB, I checked "freemem" parameter from mission planner and reducing buffer size solved my problem.

Pixhawk ArduPlaneV3.8.0beta5

LOG_FILE_BUFSIZE                  freemem                   Problem
16                                            1584                        "No IO Thread Heartbeat" 
8                                              9776                        "No IO Thread Heartbeat" 
4                                              13872                       -  

I believe that new features in ArduplaneV3.8.0beta5 uses too much memory which causes "No IO Thread Heartbeat" and "Bad Logging" problem.

OXINARF, 2017-06-07:
@ferdicakmak I doubt it is a firmware issue because several people are using Plane 3.8 beta and as well as Copter 3.5 beta without any problem. What I'm wondering is if you have gone back to Plane 3.7.1 and got good logs? You won't get the IO thread report in that version, but if you have it consistently with 3.8, you shouldn't get good logs with 3.7.1

peterbarker, 2017-06-08:
@ferdicakmak Intriguing, and great you've found something that modifies the problem for you.

I think it is entirely possible there's a memory allocation problem somewhere, so let's keep digging at this one.

I'm curious about the amount of memory you have free here.  1.5k free memory is *way* too close for comfort.

I have much more free memory:
```
RTL> Requested parameter list
APM: ArduPlane V3.8.0beta5 (b29db0aa)
APM: PX4: a6205e6e NuttX: 1a99ba58
APM: PX4v2 00360027 3435510B 38393730

RTL> Received 821 parameters
Saved 821 parameters to mav.parm
status MEMINFO
RTL> 208: MEMINFO {brkval : 0, freemem : 47696, freemem32 : 0}
```

That's with all default parameters.  Any idea why you're using up 45k more memory than my PH2 here?  Providing me a set of parameters which allowed me to replicate would be awesome.

ferdicakmak, 2017-06-08:
@peterbarker thank you for your interest in this problem. Here are the parameters I am using with Pixhawk and ArduPlaneV3.8.0beta5. LOG_FILE_BUFSIZE:16 leaves 1584 bytes, when monitored from Mission Planner.

[16X_2017-06-08_Arduplane V3.8.0beta5_mem16.zip](https://github.com/ArduPilot/ardupilot/files/1060126/16X_2017-06-08_Arduplane.V3.8.0beta5_mem16.zip)

In another test i increased LOG_FILE_BUFSIZE:32 and saw freemem:50736 but again with "No IO Thread Heartbeat".

ferdicakmak, 2017-06-08:
@OXINARF I have 2 quadplanes operational. When I was using ArduplaneV3.7.1, I didnt have logging problems. I begun to see these problems after ugrading to ArduPlaneV3.8.0beta5 consistently in both of my platforms.

peterbarker, 2017-06-08:
I tried loading your parameter file (`mem.parm`) on my Cube.

```
CIRCLE> 6: MEMINFO {brkval : 0, freemem : 40064}
After loading mem.parm:
After changing to v3:
QLAND> 205: MEMINFO {brkval : 0, freemem : 64368}
after setting LOG_FILE_BUFSIZE to 64 (and moving to mavlink2);
QLAND> 7543: MEMINFO {brkval : 0, freemem : 15216, freemem32 : 15216}
```

So... where's all your memory going?

Note that I don't have all the peripherals you might have attached to your Cube - perhaps try isolating the cube to see if one particular device is causing issues?

ferdicakmak, 2017-06-08:
@peterbarker My tests were conducted by isolating other peripherals, where only USB connected to PC.

I need to solve this issue because making LOG_FILE_BUFSIZE:4 and having log entries every 4 seconds is not 
really useful especially in vtol modes.

Do you think that disabling IMU logging would help?

peterbarker, 2017-06-10:
@ferdicakmak What is the exact type of PixHawk you're using here?

ferdicakmak, 2017-06-12:
It is original Pixhawk (PX4V2).

peterbarker, 2017-06-13:
On Mon, 12 Jun 2017, ferdicakmak wrote:

> @peterbarker It is original Pixhawk (PX4V2).

As in manufactured on behlaf of 3DR?

ferdicakmak, 2017-06-13:
@peterbarker Yes. But I am having exactly same problems with same settings when using clones too... So I dont think this is a hardware issue.

ferdicakmak, 2017-06-14:
I set LOG_FILE_BUFSIZE:4, which allows me eliminate "No IO Thread Heartbeat" problem on the ground tests. And I also performed several ground tests where logs were finely recorded at mostly 0.25Hz.

I performed a total of 6 flight with this setting without seeing "No IO Thread Heartbeat" problem. When I looked at the logs, I see lots of missing points on the data, like Voltage value below.

[image]

6 flight logs show that data is irregularly recorded.

Her is a sample log I got from my flights.

[17_2017-06-13 11-24-51_QLAND_Thermal_Shutdown.zip](https://github.com/ArduPilot/ardupilot/files/1073777/17_2017-06-13.11-24-51_QLAND_Thermal_Shutdown.zip)

So, I believe setting LOG_FILE_BUFSIZE:4 eliminates "No IO Thread Heartbeat" message, but the data recorded in flight are unusable due do irregular sampling. 

Thus, getting healty logs using Pixhawk and ArduPlaneV3.8.0beta5 in VTOL configuration still present for me.

ferdicakmak, 2017-06-14:
I am stuck with "No IO Thread Hearbeat" and "Bad Logging" problems.

I was getting "No IO Thread Hearbeat" message using settings (LOG_BITMASK: 65535, LOG_FILE_BUFFSIZE: 16) and changing LOG_FILE_BUFFSIZE to 4 didnot really solve the problem. Then I formatted sd card (FAT32) with FAT file system. It seems fine with LOG_FILE_BUFFSIZE:16 and logging is as expected. Do you think that formatting with FAT instead of FAT32 might have solved the problem? It seems to be working for now.

@peterbarker @OXINARF What should be the best settings for formatting sd cards? 

File System - NTFS, FAT, FAT32, exFAT ?
Allocation unit size - ?
Quick Format - Yes/No

peterbarker, 2017-06-14:
@ferdicakmak Nobody has sat down and worked out optimal allocation unit size.  And AFAIK you're the first to now formatting with FAT makes a difference.  @jschall might be interested in that.

briannadeau, 2017-08-09:
After upgrade to 3.8 I am seeing the message, each time it appears after the pixhawk has been up for 30 minutes.   Never saw these before 3.8

Is there something that happens at the 30 minute mark?

ferdicakmak, 2017-08-09:
@briannadeau I see these message randomly. So I don't think that something occurs after 30 minutes to cause this problem.

peterbarker, 2017-08-10:
On Wed, 9 Aug 2017, briannadeau wrote:

> After upgrade to 3.8 I am seeing the message, each time it appears after the
> pixhawk has been up for 30 minutes. Never saw these before 3.8

How reliable is that 30 minute number?

The error message indicates that some process running on the IO thread has 
blocked.  My experience to-date is that this is a call to NuttX's write() 
method isn't returning, often corresponding to a bad SD card.  These 
failures are a LOT more visible in Copter 3.8 - previously the failures 
were often silent, which wasn't good for people relying on them for their 
day-to-day work!

> Is there something that happens at the 30 minute mark?

Not on purpose!

briannadeau, 2017-08-10:
I repeated the test 3 times, just recording wall clock minutes, so it was 30 minutes +/- 1 minute.   I'm happy to provide any data that may be of help, I can reproduce it readily.  I reformatted the SD card before the test but I can try another.    The SD card is on the approved list.

avrem, 2017-08-18:
1 second heartbeat timeout seems too strict, on Navio2 setup with cheap SD card chunk writeout sometimes takes longer than that. With 5 seconds timeout logging keeps working just fine.

ferdicakmak, 2017-08-18:
I would have tried that, if it was a modifiable parameter. @tridge @peterbarker is it possible to make heartbeat timeout a variable in the .param list?

skyveyor, 2017-08-22:
Hey Guys, 

Anything I can do to help troubleshoot this problem?

I have been getting an intermittent bad logging error for nearly I year. I've swapped and reformatted multiple SD cards and that didn't fix the problem. I even swapped pixhwawk (original 3dr) flight controllers and the problem still persist...it must be something with my set up; either hardware or parameters. 

Over the past  year, for various reasons, I have used several different telemetry radios and have seen the bad logging error with each one of them. So this is an unlikely culprit. 

My primary GPS is a Septentrio AsteRx. I am backing that up with a Usmile Beitian NEO-M8N GPS. Both are working and seem to be configured properly. I normally fly with RTK. I am sending RTCM data through mission planner via telemetry.  

Lastly, I am using CAMREMOTE from VP-Systems to trigger a Sony A6000.  I have pin 55 connected to a hot shoe to get a cam trigger message. The camera gimbal is controlled by an Alexmos 8-bit card. 75% of my flights are configured this way, but I have flown with no camera, no gimbal and disconnected CAMREMOTE and still got a bad logging error; consequently, I think we can eliminate these components from the list. 

Not sure If any this relevant to the problem but want to include anything that might aid in troubleshooting. 

Please let me know if I can help, 

Dave

ranqingfa, 2017-12-24:
@skyveyor   Yes, I got the same problem too, for at least a year! Both with FW3.7.1 or 3.8.x! Also sometimes there was not any error message, and just no log after this flight but had log in the next flight, its so wired! Both plane and copter have got the same problem.

peterbarker, 2017-12-24:
On Sun, 24 Dec 2017, Leo Ran wrote:

> @skyveyor Yes, I got the same problem too, for at least a year! Both with
> FW3.7.1 or 3.8.x! Also sometimes there was not any error message, and just
> no log after this flight but had log in the next flight, its so wired! Both
> plane and copter have got the same problem.

Latest stable should be better at telling you what's wrong.  There's also 
a patch in that stops it happening so frequently.

Peter

skyveyor, 2017-12-24:
@peterbarker 

I reconfigured my quad recently. I got rid of the second GPS and added an external compass. Now I get a bad logging message the moment I arm the quad. I've long suspected this logging issue is related to using a Septentrio AstRx GNSS receiver. The pixhawk often detects the GNSS at a baud rate of 38.4 while the serial port is set to 115k per the Septentrio pixhawk integration guide.  Any chance the disparity in detected vs actual baud rate could cause this issue? 

I am using copter 3.5.3. I upgraded to 3.5.3 hoping some of the septentrio fixes would resolve the logging issue. 

Thanks and happy holidays,  

David

peterbarker, 2017-12-25:
On Sun, 24 Dec 2017, skyveyor wrote:

> I reconfigured my quad recently. I got rid of the second GPS and added an
> external compass. Now I get a bad logging message the moment I arm the
> quad. I've long suspected this logging issue is related to using a

If logging checks are enabled then it should fail to arm.  Are you saying 
it arms and *then* the logging fails?  That's... odd...

> Septentrio AstRx GNSS receiver. The pixhawk often detects the GNSS at a
> baud rate of 38.4 while the serial port is set to 115k per the Septentrio
> pixhawk integration guide. Any chance the disparity in detected vs actual
> baud rate could cause this issue?

Should be unrelated.  Could be some odd power issue...

> David

Yours,
-- 
Peter Barker                          |   Programmer,Sysadmin,Geek.
pbarker@barker.dropbear.id.au	      |   You need a bigger hammer.
:: It's a hack! Expect underscores! - Nigel Williams

ranqingfa, 2017-12-25:
@peterbarker I just did a test, armed the pilot indoor for several times and then unplug the TF card but not every time got a msg of NO IO Thread Heartbeat, also no Bad Loggind alarm!

peterbarker, 2017-12-25:
On Mon, 25 Dec 2017, Leo Ran wrote:

> @peterbarker I just did a test, armed the pilot indoor for several times
> and then unplug the TF card but not every time got a msg of NO IO Thread

OK, that much makes sense; you get to roll dice as to whether an IO 
operation returns from NuttX if you yank the SD card.  This is one of the 
reasons why we run it on a separate thread.

> Heartbeat, also no Bad Loggind alarm!

Please verify you can't arm without an SD card in there.

ranqingfa, 2017-12-25:
@peterbarker  Yes! Arm failed if no TF card

Pedals2Paddles, 2017-12-25:
On my 2.1 running 3.5 and 3.6, I usually see a `NO IO THREAD HEARTBEAT` when I first power it up. But it doesn't repeat, and everything appears to be working properly. Been seeing it for months now.  No logging errors. Arms fine.

peterbarker, 2017-12-25:
On Mon, 25 Dec 2017, Matt wrote:

> On my 2.1 running 3.5 and 3.6, I usually see a NO IO THREAD HEARTBEAT when
> I first power it up. But it doesn't repeat, and everything appears to be
> working properly. Been seeing it for months now. No logging errors. Arms
> fine.

Interesting datapoint.  If this is in a Solo it might be the OreoLEDs 
doing something screwy - I think they're on that thread too.

Peter

skyveyor, 2018-01-12:
@peterbarker 
Hello Peter,  I think you once asked for repeatable problem. Well your wish is my nightmare but will hopefully provide some insight into a vexing problem.

Below, please find a link to a tlog from mission planner from a session that had a bad logging cue. I am logging while disarmed to aid in trouble shooting. Everytime I connect to the drone I get a bad logging cue. I've also set all of the serial ports to -1, an attempt to isolate the problem. This is with a freshly formatted SD card. 

When you get a chance,  please take a look. 
 
https://www.dropbox.com/s/wj3xxu9dmvew6h9/2018-01-12%2014-54-51.tlog?dl=0

Thanks, 

David

KRCNZ, 2018-11-15:
Hello All. I just thought I'd add my observations to the mix, it may or may not help.
Yesterday I dragged 2 x Pixhawk 1's (1 x 3DR original, and 1 x CUAV clone) out from the cupboard to check out. They have both been sitting around as they have been replaced by Cubes. I have another project in mind and wanted to fire them up to see if they were operating etc. So had no peripherals attached - just the Pixhawk, SD card, and usb connection to my windows computer. I flashed both of these with the latest Arducopter Hex firmware 3.6.1. I initially used a 16GB SanDisk Ultra SD card and swapped this between each Pixhawk as I was trying it out. Both Pixhawks reported a 'No IO Thread Heartbeat' message every 30 seconds or so. I tried changing the LOG_FILE_BUFSIZE from 16 to 4 - no difference. So having read through this thread I thought about the SD card and thought it might be too big - so I installed a no name 1GB card. Still the same issue. I then reflashed one of the Pixhawks and this time changed to Chibios instead of Nuttx (worth a crack right??) - still the same consistent message on both. I then reformatted the SD card with FAT instead of FAT32 and havent neither (1 on Nuttx and the other on Chibios) have skipped a heartbeat since (see what I did there - heartbeat.....). I also put the LOG_FILE_BUFSIZE back to 16 and no problems (just touching some wood now)
Cheers Kim

jeroentaverne, 2019-06-07:
My comment is a bit late, but I noticed dropping of the one shot signals to the motors while accessing the logfiles on the SD card through Mavlink. Can this be related?

peterbarker, 2021-07-30:
This message is long gone.

closed 2021-07-30 by peterbarker