ArduPilot/ardupilot #5602: Arduplane: altitude fluctuates in my desk

aryklein, 2017-01-23:
#### Issue details
I've installed Arduplane 3.7.1 into a Pixhawk. Everything seems working fine except the altitude. The altitude fluctuates all time. My plane is stable on my desk and altimeter is not stable. It changes between -1m and 1m approximately with jumps of 10cm . Sometimes it jumps 30cm. If I lift my plane 2m (for example) the altitude marks 2m and then with the plane at 2m, the altitude goes to -0,20 or other incorrect altitude. If I open the tuning window, barometer seems working fine. Just for testing propuse I installed the Copter firmware and altitude works perfect (very stable). The problem is only with Plane firmware.
In the Plane firmware, GND_TEMP is always in 25 Celsius, and with Copter firmware, GND_TEMP is according to the environment temperature. Is that fine?
Other issue that maybe it's related with the Altitude is a message with "Bad AHRS" when I turn on my Pixhwak.
Any idea how to solve the altitude fluctuating problem?

#### Version
3.7.1

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover

#### Airframe type
Fixed Wing

#### Hardware type
Pixhwak

magicrub, 2017-01-23:
I'm not sure about the bad ahrs, but I think that is already fixed in 3.8. 

Copter grabs the home location (including alt=0) at first GPS lock and then once more when armed. Plane does it differently. While unarmed, it constantly resets home. So, if you arm the plane you should see the same behavior as copter.

Baros will always fluctuate, especially indoors.

Closing.

closed 2017-01-23 by magicrub

rmackay9, 2017-01-24:
The Bad AHRS is probably because not all the calibrations have been done (i.e. accelerometer, compass).

aryklein, 2017-01-24:
Thanks Tom for the explanation. I had not armed my plane so I thought it was a bug. Solved

magicrub, 2017-01-26:
@aryklein We've heard this question before on Plane, but not on Copter. Seems people like and expect the copter method better. We should consider switching plane to that too

magicrub, 2017-01-26:
Seems there was a recent change that on plane that is similar to this thatis causing problems: https://github.com/ArduPilot/ardupilot/issues/5623

ranqingfa, 2017-02-17:
Plane does it differently only in 3.7.1?
@magicrub

magicrub, 2017-02-17:
The altitude system was recently updated by @WickedShell and should be out on v3.8 soon

WickedShell, 2017-02-17:
@magicrub Is that being included in 3.8? I actually thought it had missed it since beta 2 has already gone out.

magicrub, 2017-02-17:
oh, that's right. i forgot the beta was out. Would be really nice to add it. Lets discuss on monday's dev call @CraigElder

ranqingfa, 2017-02-18:
My plane sometimes just fly around and cant land. It seems the baro data often drift a lot. The firmware is 3.7.1, but it looks good in 3.4.0

magicrub, 2017-02-18:
Please explain can't land?

ranqingfa, 2017-02-18:
Actually, the plane cant get into Land Mode, as usual it would. My plane flys in auto mode well, and when it ends mission, it should land, but it just flys in circle, cant land. This happens many times, but in firmware 3.4.0 never happened. It seems because the baro data drift a lot, so it does not get into Land Mode. Is there any judge if the baro  data drift l a lot,  it just flys aroud?

magicrub, 2017-02-18:
What is land mode? Can you create a new issue and post a log please?

ranqingfa, 2017-02-18:
ok! @magicrub 
 #5753