ArduPilot/ardupilot #5604: Copter: Allow Remapping of RC Motor Outputs Through Parameters

DavidIngraham, 2017-01-24:
#### Issue details
We have designed a highly integrated Pixhawk 2 Carrier Board that is suitable for use with Quads, X8, or Octo. The board has power and PWM outputs available at each corner. [(Product page for reference)](http://www.spektreworks.com/pixhawk2/multi-rotor-pixhawk21-carrier-board)

Due to the differences in motor mappings between these airframes, we need the ability to re-map the outputs to different corners of the board.

Currently, the motor mapping is hard coded in [AP_MotorsMatrix](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Motors/AP_MotorsMatrix.cpp), which pushes directly to the HAL.

This could be fixed if either of the following options implemented:

- Parameters for each output to allow arbitrary remapping
- A single parameter to switch between the traditional mappings and new mappings with consistent locations for each motor.

I will likely be able to implement this, but I would like input on which approach is most appropriate.

#### Version
Copter 3.4

#### Platform
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_
Multicopters

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_
Pixhawk 2.1

rmackay9, 2017-01-24:
I think this is already done in AC3.4.4.  All you need to do is set the RCX_FUNCTION parameters.  It's a little annoying to figure out what the parameters need to be set to but if you wanted motor1 to appear on output channel 8 you would set RC8_FUNCTION to 33.  If you wanted motor2 to appear on output channel 7 you would set RC7_FUNCTION to 34, etc.

https://github.com/ArduPilot/ardupilot/blob/Copter-3.4/libraries/RC_Channel/RC_Channel_aux.h#L67

rmackay9, 2017-01-24:
If you confirm this works, we can close this issue.

DavidIngraham, 2017-01-24:
The RCX_FUNCTION parameters are only available on non-motor outputs. I believe RC_Channel_aux lib doesn't touch the motor outputs at all: [mentioned in comment here](https://github.com/ArduPilot/ardupilot/blob/Copter-3.4/libraries/RC_Channel/RC_Channel_aux.cpp#L83). Those parameters only allow you to mirror motor outputs on non-motor channels (I'm not sure why one would want to do this).

OXINARF, 2017-01-24:
@DavidIngraham Have you tested with master? With the RC split, now you should be able to set SERVOX_FUNCTION on all outputs.

DavidIngraham, 2017-01-24:
I just re-flashed from master. I now see groups parameters for both SERVO1-16 and RC1-16, but only SERVOs have the function parameter.

OXINARF, 2017-01-24:
That's what you need, SERVO are the outputs. So you need to set SERVO**X**_FUNCTION to the motor output you want.

DavidIngraham, 2017-01-24:
Ahh, I think I get it. RC* params are now for inputs, and SERVO* params are now for outputs? That is definitely an improvement.

closed 2017-01-24 by DavidIngraham

OXINARF, 2017-01-24:
Yes, that's it.