ArduPilot/ardupilot #5606: Copter: object avoidance should support "stop" and well as "slide"

rmackay9, 2017-01-24:
Copter's current object avoidance is built on the stop-at-the-fence feature which allows the vehicle to slide along the fence barrier.  This is perhaps OK for the fence but for object avoidance it is sometime unnerving for the pilot to see the vehicle moving around as it reaches a barrier.  Instead it might be more confidence inspiring for the vehicle to simply stop.

We should add a new parameter to AC_Avoidance which allows the user to choose the behaviour, "Slide" or "Stop".

dricketts, 2017-02-03:
@rmackay9 In case it's helpful, here's the original implementation I did for the geofence, which has the stop policy instead of the slide policy: https://github.com/UCSD-PL/ardupilot/blob/geofence/libraries/AC_Avoidance/AC_Avoid.cpp.

It has some dumb ideas in it. In particular, if the fence is violated, the vehicle moves back towards the last position it held inside the fence. This is a bad idea. Instead, there should be a buffer (as in the current fence), which should be subtracted from the distance to the fence on line 30. Of course, an actual fence violation should cause a failsafe action.

I'm putting the code here instead of in a pull request because it probably doesn't build again the latest master, and I won't have time to fix it in the near future.

3dxhobbies, 2017-03-16:
Can this be added to rover as well?

rmackay9, 2017-03-16:
@3dxhobbies, I certainly hope we can add this object avoidance to rover as well.  It's a chunk of work but there's definitely some interest.  We just need to gather up the developer resources to do it.

KhushalB, 2017-10-07:
Hi, any update on this? I'm designing a collision avoidance system for a hexacopter spraying areca nuts in a dense plantation, and I need the copter to stop in place if it gets too close to the target, or leaves and branches around the copter, not glide past.

Also interested in the obstacle avoidance working not just in the direction the vehicle is travelling, but all around it, so that my copter doesn't end up damaging leaves and branches appearing from other sides. (not sure if there's already an issue on this, I will raise it if not)

rmackay9, 2018-02-13:
This is done in master and will go out with Copter-3.6

closed 2018-02-13 by rmackay9