ArduPilot/ardupilot #5607: Copter: object avoidance should work in RTL, Auto, Guided, etc

rmackay9, 2017-01-24:
Copter's object avoidance only works in Loiter, Guided mode's velocity control and AltHold.  We should extend it to work in more modes including RTL, Auto and Guided (Waypoint).

zhjgod, 2017-02-15:
When will the Auto mode use object avoidance?

felipeviveiros, 2017-10-05:
news about Auto mode use object avoidance?

lakshaydang, 2017-10-15:
Is object avoidance in auto mode is added with tera ranger tower?

X1aero, 2017-10-19:
Object avoidance using two LW20 in auto mode would be awesome...  Just sayin' : )

charleslinquist, 2018-02-07:
I realize that no one will want to commit but...
Can anyone give me an idea of when Object Avoidance will be available in AUTO mode?  If it won't be completed within the next 6 months, I'll continue working on a system that performs this function using a companion computer.  But I certainly don't want to duplicate efforts.

rmackay9, 2018-02-08:
@charleslinquist, I'm pretty sure it'll be in within 6 months.  It's not hard to make Auto mode stop before hitting things, the problem is avoid getting stuck behind things and resolving that problem requires "path planning" which is tough and could take more than 6 months.  Still, I suspect we will push in the object avoidance in Auto (RTL, Guided, etc) as a first step.

charleslinquist, 2018-02-08:
You might want to consider something that I was planning:  
When an object is detected,  
#1 brake hard (stop) 
#2 move slowly to the right and upwards at a diagonal until the object is no longer detected.
#3. continue on the up-and-right trajectory for 5 seconds
#4. move toward the next waypoint slowly for 10 seconds.
#5 if an object is detected go back to step #2 
#6 if no object is detected, resume mission.  

Not a perfect solution, because it might take quite awhile to negotiate a mountain, but  the method is unlikely to get stuck behind a building.

rmackay9, 2018-02-08:
@charleslinquist, txs for that!

mshajeehm, 2018-02-21:
waiting impatiently for this feature to be commit by SOMEONE ...

jihadrahmawan, 2018-02-28:
What is posible to read value sensor obstacle over mavlink? I'm connected sf40c to my pixhawk and i'll try to read that.. i'll make planing in my dronekit python for obstacle in guided gps waypoint mission..

peterbarker, 2018-03-01:
On Wed, 28 Feb 2018, jihadrahmawan wrote:

> What is posible to read value sensor obstacle over mavlink? I'm connected
> sf40c to my pixhawk and i'll try to read that.. i'll make planing in my
> dronekit python for obstacle in guided gps waypoint mission..

DISTANCE_SENSOR

https://github.com/ArduPilot/mavlink/blob/master/message_definitions/v1.0/common.xml#L3767

bobtitus, 2018-03-11:
I have a question : would object avoidance work when the copter does not maintain a fixed altitude?

kostkar, 2018-10-31:
The VFH or VFH* might be a good method if you have a waypoint target, in this method the vehicle would go to the direction closest to the intended direction of travel that is not obstructed. 

Advantages:
-It can navigate a complex terrain.
-avoid moving obstacles.
-work in 2d and 3d implementations.
-work regardless of sensor (more resolution of sensor just allows for tighter histograms and smoother motion).
-Can support external input (ie NN Cameras, 1d-2d-3d Lidar, Depth Cameras)

Resources:
https://en.wikipedia.org/wiki/Vector_Field_Histogram 
https://www.youtube.com/watch?v=T68MmIue0KI

Brooce, 2018-12-04:
It's been some time since last statements here and I'd like to ask what's the expected date for releasing Object Avoidance in AUTO mode. Is anyone working on it?

We will need this feature in UAVs developed by my company and possibly I'd try to implement some solution. But I don't have experience in Ardupilot development so it will take lots of time - thus I wanted to know if it's worth doing on my own. Do we know anything about the progress (@rmackay9)?

rmackay9, 2019-01-21:
It's still on the to-do list but it's not be actively worked on right now... at least I'm not currently working on it.  It's not terribly hard to make it kind-of-work.  It is a matter of changing the AC_PosControl library so that the _vel_desired.x and _vel_desired.y are shortened using the AC_Avoid::adjust_velocity() function.

Djmarck22, 2019-02-04:
> It's still on the to-do list but it's not be actively worked on right now... at least I'm not currently working on it. It's not terribly hard to make it kind-of-work. It is a matter of changing the [...]

Hi rmackay.... can I email you?

rmackay9, 2019-02-04:
@Djmarck22, sure, I'm rmackay9 at yahoo.com

highfreq, 2019-03-01:
Is there any news on this by any chance?

apryamostanov, 2019-06-10:
Dear Team,

Why not adding a Collision Failsafe?

Use case:
- Terrain: mountains
- Mode: Auto
- Situation: user wrongly set the altitude and drone collides with the mountain

If Collision risk is detected, RTL/Smart RTL could have been triggered and avoided the collision.

auturgy, 2019-06-10:
There is a terrain failsafe already, but it is linked to terrain following, not avoidance. I would expect it to work with both terrain following and avoidance enabled though.

Regards,

James

rmackay9, 2019-06-10:
@apryamostanov, yes, it's a good idea and I've been thinking about it as well.  Some of it can be done before the mission is even run because we can calculate if a target waypoint is below the terrain at that point.

apryamostanov, 2019-06-10:
@auturgy - thanks for info about terrain following.

@rmackay9 - thanks for the feedback!
The validation feature is nice! But probably more complex. It will probably require uploading of terrain data to vehicle (correct me if I'm wrong).

Are you seeing it as a pre-arm check, or it will reject uploaded mission?
Or Auto mode will not activate?

Thus the two features can be split: de-facto Failsafe and pre-validation check.

For my scenario it is more important in-flight feature (Failsafe):
- I plan route at the spot, and in my setup GC is connected to Vehicle - and none of them have internet at the moment of mission upload.
- Sometimes I fly in the desert with high dunes (even as high as 30 meters), but dunes change and thus there is no static altitude data available for desert.

rmackay9, 2019-06-10:
@apryamostanov, it might be a pre-arm check..  

By the way, we do load terrain information onto the flight controller.. this allows performing terrain altitude missions without the user of a lidar..

In any case, I can't make any promises about when this will be implemented.. support for object avoidance in Auto, RTL etc should arrive with Copter-3.7 but I suspect terrain avoidance will be later.. perhaps Copter-3.8..

apryamostanov, 2019-06-10:
Oh, that's great news about object avoidance in Copter 3.7.

That can suffice to my use case - the mountain is sheer, ~75 degrees angle - that's not normal terrain, rather an extremum/object.

Thank you very much!

charleslinquist, 2019-06-10:
For those who are interested, I have a prototype that I'm testing.  It uses a TEENSY3.5 and sits between the radio(s) and the telemetry port and uses an Ainstein US-D1 radar (~ 150' range with a 45 X 30 degree FOV).  When an object is detected by the radar, The TEENSY changes the MODE, and then does its best to fly the craft around/over the obstacle and when the obstacle is no longer seen, goes back to the original mode (which was stored).  It doesn't operate below 10M (as reported by the FC) to avoid ground reflections.  A RTL or LAND command from the ground over-rides everything. I still have a lot of "fine tuning" to do but it is starting to work. It appears likely that I'll have a package to sell before Copter 3.7 - which may make my method obsolete, we will have to see.

apryamostanov, 2019-06-10:
That's an excellent method, @charleslinquist . I was thinking of same thing.

rmackay9, 2019-06-11:
@apryamostanov, I've created a [new issue here for the terrain based avoidance feature request](https://github.com/ArduPilot/ardupilot/issues/11552).

silegc-xguard, 2021-11-05:
Hi, I'm wondering since the issue is still open... Has object avoidance been implemented in Auto mode yet ? If not, do you have a workaround to make it kinda work ?

Thanks !

rmackay9, 2021-11-05:
@silegc-xguard,

Path planning works in autonomous modes but not "simple avoidance" which is stopping.  There's some explanation here on the wiki https://ardupilot.org/copter/docs/common-object-avoidance-landing-page.html#avoidance-types

silegc-xguard, 2021-11-08:
Oh ok I see. Do you know why it hasn't been enabled yet ? Is there some implications to it or is it just a bandwith issue ?

We're trying to do collision avoidance using an OAK-D camera on an autonomous multirotor. Since the path planning aspect of the project is scheduled for later, simple object avoidance (with maybe a call to RTL on obstacle detection) would be a good milestone for us. With some insight we might be able to add it/submit a PR for it.

kaancevher, 2023-03-08:
- Trrə.

FedericoCiresolaUP, 2025-10-21:
Is there any news about this?