ArduPilot/ardupilot #5608: Copter: object avoidance with ROS

rmackay9, 2017-01-24:
Copter's current object avoidance works either through directly connecting a Lightware SF40C 360 lidar to the flight controller or by using a mavlink enabled sensor that provides the DISTANCE_SENSOR messages.  We should extend support so that we can work with ROS's SLAM point clouds

patrickpoirier51, 2018-03-28:
Randy, is this being adressed with new slam release?

rmackay9, 2018-03-28:
@patrickpoirier51, i'm afraid not yet.  What we need to do to get the object avoidance working is to somehow modify the ROS side so that it sends DISTANCE_SENSOR messages based on the distance to the closest points in the point cloud.  I actually have almost no idea how to do this.

auturgy, 2018-03-28:
@vooon would you be interested in helping with this?

vooon, 2018-03-28:
Unfortunately I don't have bandwidth, and point-cloud sources.

patrickpoirier51, 2018-03-28:
I implemented this  on my wheeled robot, its reading KINECT pointcloud and transform to 2d laser scan
http://wiki.ros.org/pointcloud_to_laserscan

bromeara, 2019-02-12:
This looks very similar to a problem I'm currently trying to solve since we are using a 3D lidar and while we are using the point cloud for processing on the companion computer I wanted to pass a couple of channels through to the autopilot for obstacle avoidance.

I was going to avoid converting from the point cloud due to latency concerns, @patrickpoirier51 I realize you are using a wheeled robot but were you able to measure the performance of your technique?

@rmackay9 My intention was originally to write a ROS node that sends DISTANCE_SENSOR messages I wasn't planning on converting from the point cloud, due to the latency concerns which may or may not be valid, however I feel like I could probably do some templating so that the source of the DISTANCE_SENSOR data doesn't matter as much. I saw the conversation in #8630 and was wondering if it would have an effect on this approach, or if that was more of a conversation on how proximity data is handled internally in general.

patrickpoirier51, 2019-02-12:
I since updated to a RealSense D435 and it’s quite performant, I used a python script to generate MAVLink avoidance but did not tried yet in a quadcopter 

I posted result in gitter vision room end November

rmackay9, 2019-02-12:
Using the point cloud info would be great if possible.  The latency might be OK if we move to a method in which we can somehow get AP to know the barrier information in "earth frame" (i.e. relative to some point instead of being relative to the body of the vehicle).  I might be possible if the time on the DISTANCE_SENSOR message was the time it started the calculations instead of the time it's sending the message.  By the way, AP also supports the OBSTACLE_DISTANCE message now so up to 72 distances can be sent at once.

bromeara, 2019-02-12:
The latency I mentioned was the time that it takes for ROS to translate a lidar packet into a point cloud and then to a mavlink message, I was planning on going from lidar packet to mavlink message. Although the earth frame would be an interesting optimization. 

You'll have to forgive me as I am still new to the ArduPilot community but when you are mentioning putting the barrier information in the "earth frame" where would this be implemented, AP_Avoidance? or would it be more of a custom AP_Proximity driver?

rmackay9, 2019-02-13:
@bromeara, the lidar packet to mavlink message should already work for an RPLidarA2 (and probably A3) using the OBSTACLE_DISTANCE message ([recent mavros PR to fix issues](https://github.com/mavlink/mavros/pull/1165)).  This is what I've been using to allow low-level obstacle avoidance to be done in the vehicle code (for Copter and Rover)

bromeara, 2019-02-15:
@rmackay9 thanks for the response, the obstacle_distance plugin looks like it will work for my initial use case. We are using Ouster's OS1 Lidar which outputs 3D point clouds that utilize  ROS' sensor_msg/PointCloud2 message, which is why I was really interested in the previously mentioned SLAM point cloud support. Was the  sensor_msg/PointCloud2 message what you were referring to? Or was it more of the sensor_msg/LaserScan message that's now supported by the obstacle_distance plugin?

rmackay9, 2019-02-15:
@bromeara, it's the LaserScan message that is now supported ([mavros code is here](https://github.com/mavlink/mavros/blob/master/mavros_extras/src/plugins/obstacle_distance.cpp#L63)).

We have some ideas/plans to maintain a simplified occupancy grid of a big list of obstacles within AP.. we just haven't quite gotten to it yet.

By the way, just want to make sure you're aware we have this [ArduPilot/VisionProjects gitter chat ](https://gitter.im/ArduPilot/VisionProjects) as well.

nkquynh98, 2019-02-22:
@rmackay9 

Firstly, thank you for your work. Now, I'm working under the project for making an obstacle avoiding drone. And I see that it can be done by the mode Alt Hold in Arducopter with sending distance_sensor Mavlink message to the Ardupilot. Therefore, I use RPLiar A2 to read the proximity distance and send it to Raspberry Pi, and then use Mavros to send the values to the DISTANCE_SENSOR message of Mavlink.

However, I can not figure how to do that. The service "/mavros/param/set" tend to send the Ardupilot-defined message (which is in here: http://ardupilot.org/copter/docs/parameters.html#rngfnd1-parameters) but not the RAW MAVLINK message. Hence, could you so me the way to use the Alt_hold with obstacle avoiding mode by mavros? Thank you.

rmackay9, 2019-02-23:
@nkquynh98,

If you're only looking to do object avoidance in AltHold with an RPLidarA2 it is also possible to directly connect the lidar to the flight controller avoiding ROS completely.  [Setup instructions are here](http://ardupilot.org/copter/docs/common-rplidar-a2.html).

Instead of using the DISTANCE_SENSOR messages, I think it's better to use the [OBSTACLE_DISTANCE](https://mavlink.io/en/messages/common.html#OBSTACLE_DISTANCE) message to send the distances to ArduPilot.  There are a couple of problems though:

- Copter-3.6.x doesn't support consuming the OBSTACLE_DISTANCE message.  This will be released with Copter-3.7 but for now you'd need to use "latest" which can be loaded using the MP's Install Firmware screen after pressing Ctrl-Q.
- mavros had some issues with it's support of OBSTACLE_DISTANCE that were only corrected a few weeks ago so you might need to use a very recent version of mavros to make it work.  The three important fixes to mavros can be seen in [mavros's commit history here](https://github.com/mavlink/mavros/commits/master) (look for 3 commits by me that start with "obstacle_distance:")

Sorry, I haven't produced documentation on how to setup the OBSTACLE_DISTANCE support but I hope to once the bits have all been released.

nkquynh98, 2019-02-23:
@rmackay9  

Oh, thank you for your supportive response. I will try connecting directly the Lidar to the Pixhawk. Also, I will use Obstacle_distance message. Btw, is there anyway to send the raw mavlink message from Mavros to FCU?

rmackay9, 2019-02-23:
@nkquynh98,

I'm not really sure about sending raw mavlink messages from mavros.  mavros is mostly a translation layer between ROS topics and mavlink messages.

AP parameters are for configuration, they're not mean to be used as a way to send or accept data from external systems.  mavlink messages are a better way to send/receive with external system.

It's probably best to ask some of these questions on the [ArduPilot VisionProjects gitter chat channel](https://gitter.im/ArduPilot/VisionProjects)..

ddomit, 2023-09-22:
Hello Everyone, 

So its almost 4 years later now. What is the recommended ROS object avoidance today?

Thank you!

HopeCollector, 2024-01-04:
@rmackay9 hello! I'm currently dev an obstacle avoiding drone based on lidar in ROS2(Humble), from the comment above I think you may need somthing that can translate sensor_msgs/PointCloud2 -> OBSTACLE_DISTANCE?

here is my project [laser_copilot](https://github.com/HopeCollector/laser_copilot), the node `obj_dist` is all you need.

there's also a simple collision prevention controller on ros side named `safe_fly_controller` in it, it can stop the drone before any 
potential collision.

currently this project is designed for px4, it you need it I can change it to ardupilot's mode