ArduPilot/ardupilot #5609: Copter: object avoidance using local occupancy grid

rmackay9, 2017-01-24:
For cases in which we cannot or do not wish to use ROS, we should be able to build up a simple occupancy grid using data collected from the proximity sensor connected to the flight controller.

This is related to this other issue: https://github.com/ArduPilot/ardupilot/issues/5608

Kiwa21, 2017-02-02:
This would be absolutely fantastic.

ayushgaud, 2018-03-20:
Can we create a custom message type for occupancy map in MAVLink using a serialized stream of the binary octree constructed from the point cloud?

rmackay9, 2019-09-26:
at the Feb 2019 developer unconference we decided that instead of an occupancy grid we should use an array of targets.   This is in master now, it's the [AP_OAObjectDatabase](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AC_Avoidance/AP_OADatabase.h).

closed 2019-09-26 by rmackay9

qleonardolp, 2022-09-16:
> at the Feb 2019 developer unconference we decided that instead of an occupancy grid we should use an array of targets. This is in master now, it's the [AP_OAObjectDatabase](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AC_Avoidance/AP_OADatabase.h).

For what reasons do you drop the implementation of a occupancy grid? Anything related with memory management/usage and dealing with a unknown size array for the grid cells?