ArduPilot/ardupilot #5610: Copter: support upward facing range finder for object avoidance

rmackay9, 2017-01-24:
Copter 3.5 supports stopping before hitting a barrier above the vehicle if an upward facing range finder can provide range measurements using the DISTANCE_SENSOR message with the orientation field set to "24".

We should extend the AP_RangeFinder class to include a new RNGFNDX_ORIENT parameter that allows the user to specify that the range finder is upward facing.  the AP_Proximity class should then accept this distance and provide it to AC_Avoidance so it can be used for avoiding barrier above the vehicle.

DavidIngraham, 2017-01-26:
Could this also be used with an arbitrary number of rangefinders in arbitrary orientations that could be used for horizontal object avoidance in AC_Avoidance?

As a cheap alternative to an SF40c, one could connect four maxbotix rangefinders over I2C.

rmackay9, 2017-01-27:
@DavidIngraham, yes, if we add this RNGFNDX_ORIENT parameter we might as well allow it to be set to a horizontal orientation as well.  The issue is that we currently only support 2 range finder objects in copter although we could easily increase that to 4.  The connection and effort to set it up will become a bit laborious and the sonar need to have unique I2C addresses.

Here's the wiki on object avoidance which you've probably already seen but just in case: http://ardupilot.org/dev/docs/code-overview-object-avoidance.html

Kiwa21, 2017-02-28:
@rmackay9  Good habits to link the wiki :) So if we can setup an external microcontroller that get the data from multiple RNGFND (say 5 : front/back/left/right/top) , we could just send DISTANCE_SENSOR messages to the FC with different orientations (0 to 7 + 24) and the code could handle it ?

rmackay9, 2017-02-28:
@Kiwa21, yes, that's right.  At least one person is working on doing something like that.  I think you'll find talk of the "POK" by Patrick Poirier in the VisionProjects gitter room: https://gitter.im/ArduPilot/ardupilot/VisionProjects

felipeviveiros, 2017-04-08:
I use on PIXHAWK sonar, like that tutorial http://ardupilot.org/copter/docs/common-rangefinder-maxbotix-analog.html , RNGFND_ORIENT is work?

felipeviveiros, 2017-04-08:
I use (XL-MaxSonar-EZ0)

rmackay9, 2017-04-08:
@felipeviveiros, yes it should.  There's a video on my (rmackay9) you-tube channel showing it working.  It's only available in AC3.5 though.
I'm going to close this issue because it's done in master and will be released soon.

closed 2017-04-08 by rmackay9

DaneSlattery, 2019-08-20:
@rmackay9 I know this is an old thread, but I can't find the PR that details this, and the docs don't mention it, but :http://ardupilot.org/copter/docs/parameters.html shows parameter options for up to 10 range finders. Is this true? 
-What is the behaviour for multiple sensors facing the same direction? 
-Are two downwards facing sensors fused in EKF2?
-As far as I can tell, sensors facing anywhere other than down are not used in the EKF2, is this true as well?

rmackay9, 2019-08-21:
@DaneSlattery, some answers below:

- in general AP doesn't use multiple lidar pointing in the same direction
- object avoidance is done in the AC_Avoidance library so it doesn't involve the EKF (which is for position estimation)
- it's true the EKF only uses downward facing rangefinders but in general we do not recommend setting the EKF's altitude source to range finder.  It's almost always a bad idea and it's not necessary for terrain following (hopefully the warnings about this are clear on the terrain following wiki pages)

hope that helps

DaneSlattery, 2019-08-25:
@rmackay9 Thank you kindly.

I've since mounted a downward LiDAR and it works wonderfully in Alt-Hold mode.

In my case, I plan on indoor flight, so I believe it to be a good use-case for setting the main EKF altitude source- considering how noisy my barometer looks.

rmackay9, 2019-08-26:
@DaneSlattery, I've found that it almost never works well to set the EKF alt source to range finder but you're welcome to give it a try.  If there's any obstructions on the ground (tables, boxes, etc) it's unlikely to work well though - it really requires a completely flat surface.