ArduPilot/ardupilot #5615: Rover: SITL gazebo long connection at initialisation

khancyr, 2017-01-25:
#### Issue details
When using gazebo sitl with rover, it seems to be an important slow down in sitl or micro disconnection 
that make launch very slow. The behaviour can be more or less long (but less that 1 min)

example ouput on mavproxy console
````
online system 1
Mode INITIALISING
fence breach
APM: APM:Rover v3.1.0 (6fc79923)
APM: Barometer calibration complete
APM: <startup_ground> Ground start
APM: Warming up ADC
APM: Initialising APM
APM: Beginning INS calibration. Do not move vehicle
Init Gyro***no link
link 1 down

APM: Ready to drive
link 1 OK
heartbeat OK
Mode MANUAL
````

#### Version
master

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[ X ] Rover

khancyr, 2017-02-16:
Update: it seems to be a problem with the gyro calibration from the gyro model in gazebo ...

khancyr, 2017-03-13:
Solve on rover by setting  <kp>100000000</kp> and  <kd>10000</kd> on wheel. It dampers the calculated collision with the ground

closed 2017-03-13 by khancyr

rajat2004, 2019-09-09:
@khancyr Could you give a few more details on how you fixed this problem? I'm having a similar problem with Airsim
I couldn't find any parameter like this in the Gazebo or rover params or even in the complete param list