ArduPilot/ardupilot #5617: Copter: ESC calibration on Pixracer can't be initiated with high throttle on startup

rmackay9, 2017-01-25:
Normally we ask users to calibrate the ESCs by doing this:

1. hold transmitter throttle high
2. restart board (board goes into ready-to-start-ESC-calibration-on-next-reboot mode)
3. restart board again (now in ESC calibration and throttle is passed through to motors)
4. wait for ESC beeps, lower throttle

Sadly step 2 fails because it doesn't see the throttle as high immediately after startup.  This is different than the behaviour of the Pixhawk1/2.

The temporary work around is for the user to set the ESC_CALIBRATION parameter to "3" and reboot the board.  on start-up it will do the "automatic calibration" which involves outputting maximum throttle for 5 seconds and then minimum throttle.

lvale, 2017-01-25:
@rmackay9 which mode are the ESC's ? I've had no issues on normal PWM, but can retest again tonight

rmackay9, 2017-01-25:
@lvale, it's actually not ESC related.  It's not even getting into the ESC calibration mode because it's not recognising that the throttle is held high on startup.

lvale, 2017-01-25:
@rmackay9 ok, I'll test tonight with logic analyser attached to mot_outputs and if I can manage the wiring also to RCIN

rmackay9, 2017-02-15:
This is fixed in master and has gone out with -rc1.

closed 2017-02-15 by rmackay9