ArduPilot/ardupilot #5618: DO_CHANGE_SPEED for vertical speed

juandiazb, 2017-01-25:
## Issue details
I have been encountering a problem for long now in which I need the copter to do a vertical sweep in AUTO mode very slowly. For that, I have lowered the parameter WPNAV_SPEED_DN, but then it takes a dangerous long time for the copter to descend when navigating through the rest of WPs.
The enhancement I am proposing is that either vertical speed can also be changed for a given set of WPs using the mission command "DO_CHANGE_SPEED" or that a new command is created for this.

#### Version
Latest

#### Platform
Copter

#### Airframe type
All

#### Hardware type
Pixhawk

#### Logs
N/A

rmackay9, 2017-01-25:
Yes, a very good idea.

ChrisBird, 2017-08-29:
Wouldn't http://ardupilot.org/copter/docs/common-mavlink-mission-command-messages-mav_cmd.html#mav-cmd-condition-change-alt do this already?  I noticed it seems to be in Mission Planner as "Continue and Change Alt"......

Param1 should be the ascent and descent rate while param7 is the actual alt desired.  I'd recommend the normal rate be set and then for the tasks that require the very slow descent rate then this is used?  I'll do a SITL test tomorrow and see if it behaves as I'm thinking it will.

peterbarker, 2017-12-19:
@ChrisBird How did your investigations go?

ChrisBird, 2017-12-26:
I'll rerun it later tonight but I think I got an unsupported mission response.  I'll double check and if so I'll fix it up so that it takes it into effect and do a pull request.

ChrisBird, 2017-12-26:
So I updated the library to take the new message, that works well.  Now the interesting issue, I thought I'd use the xy speed changes as a template for this change.  That then made me rethink my approach to this issue.

The xy speed changes work by updating the xy speed (WPNAV_SPEED) parameter runtime value to whatever is set in the mission.  So I thought why not update the Z value but that is an issue in that which value do I update as we have an UP and a DN value.  The Mavlink message only contains 1 parameter which is a rate.

So some options (possibly more but these are my initial thoughts):
1: Set the WPNAV_SPEED_UP and DN to the new value of rate.  _Pros: Not a lot of coding changes required.  Cons: We have asymmetrical values for a reason......_
2. Look at the new value, if positive then update UP and if negative update DN. _Pros:  Not a lot of coding changes required.  Cons: You need two commands to update the ascent and descent rates.  Not sure if GCS's other than mission planner will allow negative values on the rate parameter._
3. Use new value for relevant direction but don't update the runtime value of the parameter.  Then update all areas that use the current values.  At the next waypoint have it reset the values back to the parameter values.  _Pros:  Wont overwrite the parameter values, would automatically revert after next waypoint (technically 2 along - needs to use the new values between the next waypoint and the following one).  Cons:  Lots of code changes, more testing, not sure its the behavior we actually want._
4. Update mavlink msg or create a new mavlink msg with the asymmetrical values.  Would then update the runtime values with the new values.  _Pros: Would have two values to maintain our asymmetrical parameters, cleanest solution as far as i can see.  Cons: Additional coding, not sure if we can extend the mavlink message (may have to use a new message or custom message) - I havent changed anything in the mavlink space so will take me longer (but happy to do).  Would require GCS's to be updated with change in mavlink message_

@rmackay9 @peterbarker can you guys provide some feedback as to what you think I should do on this.  Happy to go with the flow :-)  But I am leaning towards option 4 (with extending existing msg type)

ChrisBird, 2018-01-17:
Since ive not received any feedback I'll go with option 4...

ChrisBird, 2018-06-15:
So after a not touching it for 4 months, I decided to put some effort back into it the last couple of days. I've settled on adding an addition message type, MAV_CMD_DO_CHANGE_VSPEED.

It has involved updating the Mavlink module (to add a new message type - added it to the ardupilotmega.xml), the AP_MIssion library and the Copter auto code.  Still ironing out a few bugs and I expect that it'll be ready for me to generate the PR's by end of next next week.

ChrisBird, 2018-10-04:
I must be missing something obvious in the code, it appears to overwrite the value.  Havent been able to work this one out, so if someone else wants to take a crack.  If not then I'll revisit it later.

chobitsfan, 2018-10-09:
hi @ChrisBird I also need this function in my research project. I use different approach. When adding DO_CHANGE_SPEED to a mission. set param1 to 2 for changing climb speed. set param1 to 3 for changing descent speed.

ChrisBird, 2018-10-13:
@chobitsfan i saw the pr for it or some of it. Feel free to get this one across the line if you need it.  I had to work on some other items as they were items i needed.

khancyr, 2019-05-31:
closed by https://github.com/ArduPilot/ardupilot/pull/9535

closed 2019-05-31 by khancyr