ArduPilot/ardupilot #5620: Stop using -fsingle-precision-constant

lucasdemarchi, 2017-01-25:
This is the usual case in which the remedy creates more problems than it's solving. By passing `-fsingle-precision-constant` to the compiler we make literals in the code to be interpreted as float rather than double. The problem is that there's no way to tell that a literal is in fact double so in cases in which it's needed, it's a) truncated to float and b) cause a warning of implicit float to double promotion when involved in other operations. This is what happens today on `libraries/AP_Math/location.cpp`.

It's also the case of least surprise in which C++ developers already expect that to be true. It's very easy to learn for those who don't know, makes our code "less special" and the warning `-Wdouble-promotion` is there to help mitigate the problems.

Plan to solve it:

1. Remove that flag from the compiler
1. Fix the warnings throughout the codebase in cases a double literal is used when it should be using a float one

With this in place we should be able to eliminate the remaining implicit float to double conversions.

khancyr, 2017-01-25:
nice idea ! I see those warning (mostly in gps and location). Another solution could be to make them static const double instead of define !

lucasdemarchi, 2017-01-25:
@khancyr yes, I agree. But it's just a matter of time until more appear.  I don't like these surprises because we have a flag to deviate from the standard.  Also, this makes the behavior different on sitl/linux than what it is on stm32 boards.

rmackay9, 2017-01-26:
I'll ask @CraigElder to add to the agenda for the next dev call?

CraigElder, 2017-01-26:
Thanks Randy. Added

lucasdemarchi, 2017-01-31:
As discussed today on the dev call feedback was that this is not desired because would need to change the codebase to add the f suffix and new people get it wrong which would cause operations on double variables rather than float.

closed 2017-01-31 by lucasdemarchi