ArduPilot/ardupilot #5623: Plane: Home altitude is no longer reset while disarmed

WickedShell, 2017-01-26:
#### Issue details
Not resetting the altitude while disarmed is a major problem for EK2_ALT_SOURCE 2 users as the initial altitude will fluctuate quite a lot on the first GPS fix. This is particularly awful for users who's aircraft is working on gaining a RTK fix.

The scope of the error is such that I'm sitting here with an aircraft that is reporting a perfect 9 meters high, even though its on the ground, and it has been very stable there after gaining a good fix, but because the initial fix was a bit off it won't reset the altitude/home spot, which compromises the use of the aircraft for mapping quite significantly.

I understand the reasons why this went in that are outback challenge/landing/taking off from somewhere other then the initial power location related, but this needs to be a parameter controlled behavior, and I feel pretty strongly the correct default behavior is to revert to the old style of resetting altitude.

#### Version
Master

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover

#### Hardware type
EK2_ALT_SOURCE 2 with both u-blox and septentrio units.

ranqingfa, 2017-02-17:
You mean the firmware 3.7.1? And altitude is reset when disarmed  before?
@WickedShell

WickedShell, 2017-02-17:
Fixed in #5647.

@ranqingfa When home moved it wasn't resetting your relative altitude as referenced from the home waypoint correctly. This was a major problem if you were using EK2_ALT_SOURCE 2, but less of a problem if your GPS wasn't your primary altitude source.

closed 2017-02-17 by WickedShell