ArduPilot/ardupilot #5628: UAVCAN broken on Pixhawk 1

hsteinhaus, 2017-01-26:
#### Issue details
With https://github.com/ArduPilot/ardupilot/commit/29bc07f410df065d276d4205d735c2643eee1ea8 , UAVCAN support has become unavailable on any Pixhawk 1 (px4-v2) based board (which is probably 90% of all ArduPilot boards). This change basically bricks any Pixhawk1-based drone equipped with UAVCAN sensors and/or actors, cutting the link to the ESCs and/or to the GPS (plus some payloads). Re-enabling UAVCAN it is rather hard, as there is a limit enforced by the build control scripts, that prevent generating firmware images larger than 1M to guard against a silicon erratum of early Pixhawk 1 boards (even if the destination board is not affected by this bug). Current ArduCopter builds are 3.5-dev are just above this limit with UAVCAN enabled.

As a result, no UAVCAN-enabled Pixhawk1-based drone can ever be updated to ArduCopter 3.5 until this issue is fixed (or the Pixhawk has been replaced by a currently basically unavailable Pixhawk2). The same probably applies to fixed wing and ground-based vehicles for their appropriate versions.

#### Version
master (tested: ArduCopter 3.5-dev)

#### Platform
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover

#### Hardware type
All px4-v2-based boards.

@pavel-kirienko FYI

proficnc, 2017-01-26:
Pixhawk 2 is not unavailable, the next batch start delivery in feb.  

Regarding PH1's, many can run fmuv3

tridge, 2017-01-26:
hi Holger, its not unavailable, you just need to load fmu-v3 not fmu-v2. Unfortunately the GCS loaders haven't caught up yet with automatically loading the right version.
Please try loading the px4-v3 firmwares manually and you should find it works for you. Then we need to work with Michael Oborne, Bill Bonney and Don Gagne to update the firmware load screens.
Note that the current builds of copter and rover are below 1M for px4-v3, so those can get uavcan even on boards with the old silicon problem

tridge, 2017-01-26:
btw, if uavcan does still fit below 1M for copter when we do the 3.5 release then I think we should enable it in that build for px4-v2 too

davidbuzz, 2017-01-27:
so px4-v2 = older cpus with 1Mb errata and px4-v3 = newer cpus with no problems going to 2Mb?

proficnc, 2017-01-27:
Correct

tridge, 2017-01-27:
Randy and I are trying to see if we can squeeze the 3.5 release onto the 1M boards with uavcan enabled

tridge, 2017-01-27:
I have a PR pending to fix this in #5633

tridge, 2017-01-29:
uavcan is now enabled on px4-v2 again in master. We have removed a number of other drives instead

hsteinhaus, 2017-01-29:
Just gave the latest master a try on the bench - UAVCAN working with Copter as expected. Thanks for the quick fix!

closed 2017-01-29 by hsteinhaus