ArduPilot/ardupilot #5631: Error building copter with waf

DaveBGld, 2017-01-26:
#### Issue details

Working on Bash on Ubuntu on WIndows 10, the GitHub clone on Windows folder C:/D/Ardupilot

When running   **# waf copter**    I get the following result

`dbg@DESKTOP-U25RNHO:/mnt/c/D/Ardupilot$ waf copter
Waf: Entering directory /mnt/c/D/Ardupilot/build/bebop
Waf: Leaving directory /mnt/c/D/Ardupilot/build/bebop
Build failed
Traceback (most recent call last):
  File "/mnt/c/D/Ardupilot/modules/waf/waflib/Runner.py", line 270, in task_status
    return tsk.runnable_status()
  File "Tools/ardupilotwaf/git_submodule.py", line 75, in runnable_status
    out = self.generator.bld.cmd_and_log(cmd, quiet=Context.BOTH, cwd=self.cwd)
  File "/mnt/c/D/Ardupilot/modules/waf/waflib/Context.py", line 453, in cmd_and_log
    raise e
WafError: Command ('/usr/bin/git', 'submodule', 'status', '--recursive', '--', '/mnt/c/D/Ardupilot/modules/mavlink') returned 1`

This after successfully running   **# waf configure --board bebop --static**

`dbg@DESKTOP-U25RNHO:/mnt/c/D/Ardupilot$ waf configure --board bebop --static
Setting top to                           : /mnt/c/D/Ardupilot
Setting out to                           : /mnt/c/D/Ardupilot/build
Autoconfiguration                        : enabled
Using static linking                     : yes
Setting board to                         : bebop
Checking for program 'arm-linux-gnueabihf-ar' : /usr/bin/arm-linux-gnueabihf-ar
Using toolchain                               : arm-linux-gnueabihf
Checking for 'g++' (C++ compiler)             : /usr/bin/arm-linux-gnueabihf-g++
Checking for 'gcc' (C compiler)               : /usr/bin/arm-linux-gnueabihf-gcc
Checking for program 'arm-linux-gnueabihf-pkg-config' : /usr/bin/arm-linux-gnueabihf-pkg-config
Checking for need to link with librt                  : not necessary
Checking for 'lttng-ust':                             : disabled for static build
Checking for 'libdl':                                 : disabled for static build
Checking for 'libiio':                                : disabled for static build
Checking for HAVE_CMATH_ISFINITE                      : yes
Checking for HAVE_CMATH_ISINF                         : yes
Checking for HAVE_CMATH_ISNAN                         : yes
Checking for NEED_CMATH_ISFINITE_STD_NAMESPACE        : yes
Checking for NEED_CMATH_ISINF_STD_NAMESPACE           : no
Checking for NEED_CMATH_ISNAN_STD_NAMESPACE           : no
Checking for header endian.h                          : yes
Checking for header byteswap.h                        : yes
Checking for program 'python'                         : /usr/bin/python
Checking for python version >= 2.7.0                  : 2.7.6
Source is git repository                              : yes
Update submodules                                     : yes
Checking for program 'git'                            : /usr/bin/git
Gtest                                                 : statically linked tests not supported
Checking for program 'arm-linux-gnueabihf-size'       : /usr/bin/arm-linux-gnueabihf-size
Benchmarks                                            : disabled
Unit tests                                            : disabled
'configure' finished successfully (3.120s)`

> 

#### Version
_What version was the issue encountered with_

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_

#### Logs
_Please provide a link to any relevant logs that show the issue_

OXINARF, 2017-01-27:
This is a development support question and those should be posted at the forum (http://discuss.ardupilot.org) or asked about in Gitter (http://gitter.im/ArduPilot/ardupilot).

Since you have already opened this, I'll try to help. What happens when you run `git submodule  status --recursive  -- /mnt/c/D/Ardupilot/modules/mavlink`?

DaveBGld, 2017-01-27:
Thanks for your reply. I found a solution.

The problem stems from my desire to develop for Bebop2 and the required arm libraries available on Linux, plus the fact that despite I have many years experience developing on Windows, I know nothing about Linux. 

So, right now I am using Bash on Ubuntu on Windows (BUW), cloned the ardupilot GIT to a mounted windows folder from within BUW, and made an alias to waf-light there so I can run your included version of waf, and then I'll get started developing on Eclipse on windows and continue making the project under  BUW

closed 2017-01-28 by OXINARF

lucasdemarchi, 2017-01-31:
there's a waf binary on tree. You can use it instead of your on alias.

In Linux  the current directory is usually not on PATH so to call a command from the current directory you do: `./waf`, not `waf`.