ArduPilot/ardupilot #5638: AC3.4.4 SF11/C Serial Issue

roque-canales, 2017-01-27:
Hello, Attached you will find *.bin regarding issue.

SF11/C is connected to serial port to drone but not used during this flight.
After auto-mission using barometer, I switched to RTL.
When drone come vertical point of takeoff point, it started descend à 200cm/s as tuned.
BUT it landed at this speed.
It not slowed to 50cm/s for final toutch-down as tuned.

Hopefully, I dont break anything.

Regarding log we can see SF11/C glitch from AGL alt to 0.....
I think that even if we don't use SF11/C, but if it's connected, AC3.4.4 set the offset with baro home alt following SF11/C measurements.

And.. in this case SF11/C sending bad alt data, AC3.4.4 continue using last offset home alt.

For the next release it would be great that EKF dont use offset to home alt if SF11/C is sending bad data.

Also, I think that issue for bad measurement are linked with bad serial driver for SF11/C....
I tell that because, SF11/C work very well when connected directly to laptop, it never fail....
When connected to ac3.4.4, it make spikes, and seems to be unstable.
https://guides.github.com/features/mastering-markdown/
[2017-01-27 13-01-23.zip](https://github.com/ArduPilot/ardupilot/files/735558/2017-01-27.13-01-23.zip)

magicrub, 2017-01-27:
> I tell that because, SF11/C work very well when connected directly to laptop, it never fail.... When connected to ac3.4.4, it make spikes, and seems to be unstable.

Are you saying that when you look at things while connected laptop, presumably short-range and stable targets, it works fine but not as well when used in a real-world environment and very long ranges of >50m where the data is not as clean and stable?

roque-canales, 2017-01-28:
Ok Magicrub that's true.

Regarding EK2, why filter continue using out of range sf11/c output (sf11/c output 130m when out of range). I think that is the main issue..... because filtering this input value, destabilize EK2.

Normal I thinks that out of range value must be rejected by filter instead of be input and used by EK2.
When sf11/c output 130m this generate: 
ERROR TERRAIN ALT VARIANCE
ERROR OS VERT VARIANCE

Also, how to explain Baro offset occured during landing while sf11/c is working well? this cause crash (landing at 2m/s it's  a crash)

magicrub, 2017-01-29:
Range values of 130m Should be rejected by your rangefinder Max range Parameter. Have you set it? It should be a value lower than the max range of your list.. around 75℅.

roque-canales, 2017-01-29:
Max range was set to 12000cm. So EK2 still using 130 meters readings.

So following logs and print screen on the top of this post why barometer (in green) was offeseted by 30 meters during final descent?

This cause crash...

magicrub, 2017-01-29:
Advertised max range is never reality. Please reduce that. I suggest 100m
or 80m. The wiki stars this.

roque-canales, 2017-02-01:
Ok I will test it.

Juzzle1, 2017-02-10:
I'm also using the SF11/C with AC 3.4.4 and haven't seen the issues you're describing, but changing some settings on the SF11/C could help get rid of momentary out of range signals which would prevent the EKF from unnecesarily flipping between baro and lidar readings over a short time period.

From the SF11/C documentation (Page 12 http://lightware.co.za/shop2017/index.php?controller=attachment&id_attachment=11):

Setting option N to 'hold last result' and option P around 1 or 2 seconds would mean that if there is a temporary glitch with the SF11/C, it continues outputting the last good value for a second or two, giving it enough time for it to get good measurements again.  If it doesn't, it outputs 130m out of range signal as before.

n: Output on lost signal (130.00 m | hold last result)
p: Confirm lost signal (0.0 ... 120.0 seconds)

roque-canales, 2017-02-12:
Thank you Juzzle1, this is good feature that can limit impact on ek2 when range finder is out off range.
But only for short period 2-3s max.

For 15s or more out of range value ek2 is badly impacted, as you can see on my logs.

Does any body can explain why EK2 AGL Offset as jumped at the end of flight causing drone crashing on the ground at max wp down speed?

Juzzle1, 2017-02-12:
Looking at the logs, it seems the problem occurs much earlier in flight - the LIDAR altitude does not correlate with baro / GPS altitude even on the initial climb and seems to top out at around 50m even though both GPS and baro indicate the copter is still climbing. 

Towards the end of the flight, when the 'glitches' occur, you can see the  SALT gradually increase between the glitches until it tracks the GPS and baro for the final descent. 

Perhaps a faulty SF11/C? 

Probably best if one of the devs could take a look at this.

[image]

roque-canales, 2017-02-13:
Ground is not plane below.

From the rtl start ground drop  by 30 meter, so it's normal

Question is why reference alt, jump at the end of flight?

Juzzle1, 2017-02-13:
Seems this was altitude reset was related to the EKF switching lanes during the descent.

I have a theory why this might have happened, but I could be completely wrong.

1) During approach to horizontal RTL position, ground is sloping downwards
2) Rangefinder stops providing good values, and BARO alt stays the same
3) When the rangefinder starts providing good values again, there is a big offset between this and the EKF altitude estimation.

[image]

Juzzle1, 2017-02-13:
With sloping ground and with the copter angle caused by forward flight etc, you might find the distance the rangefinder has to measure is MUCH greater than the actual AGL. 

As magicrub suggests, lower your RNGFND_MAX_CM. For your application I would suggest going as low as 5000 (50m)

roque-canales, 2017-02-13:
Ok, thank you Juzzle.

peterbarker, 2021-07-30:
This looks to have been resolved years ago.

closed 2021-07-30 by peterbarker