ArduPilot/ardupilot #5645: Rover: skid-steering in reverse bug

khancyr, 2017-01-30:
#### Issue details
See http://discuss.ardupilot.org/t/skid-steering-in-reverse/14112/9

Resume : in skid steering mod the rover can get stuck when use do_reverse

#### Version
To reproduce the bug on sitl, make a line of way point, put a do_reverse between to waypoint, launch the mission. The rover will get stuck on spot after the rotation. It also rightfully trigger a fs_crash !

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover

gmorph, 2017-02-09:
Confirmed this is broken in current master - not just the 3.1.1 release.

gmorph, 2017-11-29:
Is this still a bug?  The new master changes should of fixed this?

rmackay9, 2017-11-29:
It's better than before but I've still seen it get stuck.