ArduPilot/ardupilot #5658: Analog Airspeed doesn't work

stundenblume, 2017-02-02:
#### Issue details
After updating from beta1 to beta2 the Airspeed on pin 15 doesn't work. After uploading beta1 it works again. But after uploading beta2 it still don't work
#### Version
3.8.0beta2
#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x] Plane
[  ] Rover

#### Airframe type
VTOL, Quad, H-Frame
#### Hardware type
Pixhawk
#### Logs
I will provide link to log later.

tridge, 2017-02-03:
hi @stundenblume, can you check what ARSPD_TYPE is set to? We changed from using ARSPD_ENABLE to ARSPD_TYPE. It should have automatically set ARSPD_TYPE according to your old settings, but maybe that failed?

stundenblume, 2017-02-03:
Hi @tridge, It was set to 1. I noticed the new parameter but couldn't find any information about it and tried also 0 but without noticing any change.
Here is a log file: http://www.file-upload.net/download-12285224/2017-02-0214-53-16.bin.zip.html (~7MB)

OXINARF, 2017-02-03:
@stundenblume If it's an analog airspeed the parameter should be set to 2, here is the description: http://ardupilot.org/plane/docs/parameters.html#arspd-type-airspeed-type
When you update to beta2 do you start with clean parameters? If you aren't, what was your ARSPD_PIN parameter value in beta1?

stundenblume, 2017-02-03:
@OXINARF 
> If it's an analog airspeed the parameter should be set to 2

OK, I will try it.

> When you update to beta2 do you start with clean parameters?

No

> If you aren't, what was your ARSPD_PIN parameter value in beta1?

it was 15 and it's the same in beta2

OXINARF, 2017-02-03:
Ok, I saw why the conversion code doesn't work.

@tridge The conversion code is doing a set default for the type parameter. But a set default will only set a value if not configured (which will be configured to 1 for people using an airspeed in previous versions). I think the code should be:

```
if (_type.load() && _type.get() == 1 && _pin.load() && _pin.get() != 65) {
    _type.set_and_save(TYPE_ANALOG);
}
```

The issue with this is that if you use an analog sensor and then change to a MS4525D0 you will always get changed to analog unless you change the pin to 65.

stundenblume, 2017-02-03:
> an analog airspeed the parameter should be set to 2

that's the solution. Thanks for your help @OXINARF !

Should I close or rename this issue?

OXINARF, 2017-02-04:
I've renamed. Thanks!

magicrub, 2017-05-29:
Looks like this fix was never actually turned into a PR.

tridge, 2017-05-29:
unfortunately that fix doesn't work. 

Imagine this scenario. A user had an analog sensor with plane 3.7.1. Then they upgrade to 3.8.0. With this suggested change the upgrade goes great, and it works with an analog sensor.
Then a month later the user decides to get a digital sensor. So they buy and install an i2c sensor, and set ARSPD_TYPE=1 to configure the digital sensor. Then this suggested code runs:

if (_type.load() && _type.get() == 1 && _pin.load() && _pin.get() != 65) {
    _type.set_and_save(TYPE_ANALOG);
}

it will change ARSPD_TYPE back to 2 for analog! They will get extremely frustrated as no matter how hard they try they can never convert to a digital sensor.
It is a surprisingly tricky problem to get right.

tridge, 2017-05-29:
btw, the reason I coded it the the way it is in master is that at least with it erring to be an i2c sensor the autopilot can detect the error and give an arming error and display an error on the GCS. We can tell if an i2c sensor is not present.
If we coded it differently and tried to make it analog then it will always show healthy, as we have no way to detect an unhealthy analog sensor.

OXINARF, 2017-05-29:
@tridge They can change to digital. But yes, there is an issue and I wrote about that:

> The issue with this is that if you use an analog sensor and then change to a MS4525D0 you will always get changed to analog unless you change the pin to 65.

My only issue with current code is that it doesn't work if the user ever changed the old enable parameter (now called type). I agree it isn't an easy upgrade path and I'll have to defer to you on what to do here.

magicrub, 2017-05-30:
Can we be more smart about this and make an exception if we've detected it in the i2c bus?

OXINARF, 2017-05-30:
I don't think we can. If I remember my discussion with @tridge in the PR that introduced the type parameter the problem is that the I2C airsensor share the address with baro sensors and doesn't have any distinguishable ID to be asked for. This means we can't autodetect if the sensor present in an I2C bus is an airspeed sensor - if we could do that we could even get ride of the type parameter.

tridge, 2017-06-07:
i think the current code is the safest option, even though its not ideal.

closed 2017-06-07 by tridge