ArduPilot/ardupilot #5660: Arduplane RPM logging only works for non-zero values

Georacer, 2017-02-03:
#### Issue details
The RPM sensing (at least the RPM1 measurement) works ok when a non-zero RPM signal is fed onto Pixhawk.
However, when the propeller is stopped, instead of measuring 0 RPM, there simply is no measurement. 
This gets worse, as Mission Planner displays the same running value if no updated value is received, so if your engine dies mid-air, you will never get a 0 RPM reading on the GCS.

A demo plot from a .tlog is shown below. Notice how there are samples (triangles) only for non-zero measurements. When the RPM is 0, there is simply no data point.
[image]

I could not find an RPM reading in my .BIN log, perhaps I need to turn on some logging flag. I'm happy to reproduce the test if needed.

#### Version
ArduPlane 3.7.1

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover

#### Airframe type
Pilatus Porter with DLA 32cc IC engine and an optocoupled RPM output.

#### Hardware type
Pixhawk 1

#### Logs
[rpm.zip](https://github.com/ArduPilot/ardupilot/files/750778/rpm.zip)

magicrub, 2017-02-04:
I'll take a look

Georacer, 2017-02-21:
Should I "bump" this while it's warm?

magicrub, 2017-02-21:
I actually looked at this today and then got side tracked. I'll do it now. Sorry for the delay!

Georacer, 2017-02-21:
Hey, no problem! It's not like I paid you for it or anything ;)

magicrub, 2017-02-21:
confirmed, if you stop getting signals, or get them slower than 1 Hz, then it sets the "quality" to zero and the healthy goes false and it will no longer log it. So, values of zero can never be logged. I'll (or someone else?) have to figure out a way to fix it later, but for now a workaround is to set RPM_MIN_QUAL=0. That forces it to be healthy all the time.

Georacer, 2017-02-21:
Is there an actual logged value (albeit unhealthy) if you stop getting pulses? What is its value?

Georacer, 2017-02-21:
A friend helped me realize that RPM_MIN_QUAL is a parameter, not a program variable. I'll test it and answer the question myself.

magicrub, 2017-02-21:
fixed via https://github.com/ArduPilot/ardupilot/commit/67486dc3f0f82fef8077494ebcf6dd4ffdedc4f1 by always logging when enabled instead of when healthy. After fixing it I went to check on copter to fix it there too and it was already doing the enabled check!

closed 2017-02-21 by magicrub

Georacer, 2017-02-21:
Wouldn't it be better if the rpm measurements were also sent via mavlink, as well as logged? I.e. do the same with the GCS library.

magicrub, 2017-02-21:
yes, we should do that too.. just like copter! Fixed via https://github.com/ArduPilot/ardupilot/commit/69a10042eb8234ed5b55075cf9e33e1ab85025c9

Thanks @Georacer !

Georacer, 2017-02-21:
Thank **you**