ArduPilot/ardupilot #5664: support DJI NAZA GPS+Compass

geofrancis, 2017-02-04:
the dji naza gps/compass puts both over a single serial link not requiring the use of the i2c bus, if it was supported it would mean we could use dual external compass easily since there would be no i2c conflict.

inav have it working so it might be a good place to start
https://github.com/iNavFlight/inav/wiki/GPS--and-Compass-setup

proficnc, 2017-02-05:
If you look at this, they are using I2C for both GPS and the compass.

This is a terrible idea!   It's bad enough that we still use i2c for the compass. 

The traffic we need on the ubx data for the gps would flood the i2c bus.

As far as cheap GPS's... define expensive? The Here is $48 with M8N, correct flash, battery backed ram, status LEDs, compass, and additional Saw filter /LNA 

There are many cheap GPS's that use the correct serial port, and i2c that are even cheaper. 

So I struggle to see how reducing the capability of the system is an improvement

OXINARF, 2017-02-05:
@proficnc According to the inav documentation (and code looks like it too) this uses serial for both GPS and compass, not I2C.

I agree though that unless you want a fake/flawed GPS, it's harder to go cheaper than what we have already.

proficnc, 2017-02-05:
Cool, looking at it yes, it looks like NAZA does... but it looks like that's NMEA.. which again is a regression

geofrancis, 2017-02-05:
yea the point is that we can use the compass without i2c, doesnt arducopter still use nema for rtk gps units anyway?

OXINARF, 2017-02-05:
> Cool, looking at it yes, it looks like NAZA does... but it looks like that's NMEA.. which again is a regression

It doesn't look like NMEA, it looks like a custom protocol with a single message with all the information we need.

>  with 3 compass's including internal you can get proper reduncancy

How does I2C prevent you from having redundancy?

> doesnt arducopter still use nmea for rtk gps units anyway?

Did it ever? The NMEA parser doesn't even support RTK fix **status**, all the other GPS drivers (that have RTK) do.

To conclude, I don't think this will be a priority for anyone in the dev team, but PRs are welcomed.

dhackens, 2017-02-05:
HelloI think you have me confused with someone else .... I did not ask this question ?
Dan H.

OXINARF, 2017-02-05:
@dhackens Nobody quoted you. Like your e-mail says you got a notification because you subscribed to this issue.

geofrancis, 2017-02-05:
@OXINARF the i2c stops you plugging in more than 1 i2c compass or you get address conflicts, as far as i know there are no other compass's supported other than derivatives of the 5883l that all use the same address.

proficnc, 2017-02-06:
There are heaps of compasses supported, in fact the Here+ has 4 I2C compasses on board, and you can plug two of them into a pixhawk 2 with no conflicts.

We support the BMM150, hmc5983, LIS3MDL, LSM303D, AK series, MPU9250...

geofrancis, 2017-02-06:
but you find any of them combined with a m8n.

rmackay9, 2017-02-06:
This seems like a valid request.  We generally try to support a wide variety of hardware.  I have a pretty full plate so it's not likely that I'll get to this soon but perhaps someone else will have a look so it's fine for it to be here as a to-do item.

magicrub, 2017-02-18:
My friend @Gregorydz is wanting to work on this. Has anyone else started? If so, would they like to collaborate? Otherwise, he'll be starting on it soon.

geofrancis, 2017-03-10:
I can help with testing.

magicrub, 2017-03-10:
@gregorydz have you had a chance to take a look at this?

Gregorydz, 2017-03-11:
No unfortunately I have not yet, been bogged down by school stuff.

sincoder, 2018-01-11:
waiting for this!  connect Compass through TTL will be more stable than I2C

geofrancis, 2018-05-20:
@night-ghost has done the code for this but it would need to be merged into the main ardupilot. @rmackay9

IamPete1, 2021-08-07:
Is there still interest in this? There a loads of better options available now.

sin5678, 2021-08-18:
no much naza gps now

rmackay9, 2021-08-19:
I guess we have consensus to close this..

closed 2021-08-19 by rmackay9

intelligentpotato, 2023-05-04:
I would very much love to see NAZA GPS support in ArduPilot. Especially since it was dropped in iNav 6.