ArduPilot/ardupilot #5666: SITL on Windows fails with non-standard MAVProxy install location

stephendade, 2017-02-05:
#### Issue details
Running SITL under Windows, sim_vehicle.py assumes that MAVProxy is installed in `C:\Program Files (x86)\MAVProxy`. If a user has installed MAVProxy in a different location, SITL will terminate with the following error message:

`Unable to start 'C:\Program Files (x86)\MAVProxy\mavproxy.exe': The specified file was not found.`

This directory appears to be hard-coded in https://github.com/ArduPilot/ardupilot/blob/master/Tools/autotest/sim_vehicle.py#L651

Suggested fix would be either a message telling the user that MAVProxy is required to be installed in `C:\Program Files (x86)\MAVProxy` or the ability to read the install location of MAVProxy from the Windows registry (appears to be HKEY_LOCAL_MACHINE\SOFTWARE\WOW6432Node\Microsoft\Windows\CurrentVersion\Uninstall\{D81B9EDA-1357-462E-96E4-B47372709F7C}_is1\InstallLocation) 

#### Version
Current master branch in Github

#### Platform
[ X ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover

peterbarker, 2017-02-06:
@stephendade can we have the installer add the mavproxy install directory to %PATH%?  That would allow us to invoke mavproxy.exe without specifying an absolute path.

stephendade, 2017-02-06:
Yep, that can work ... I'll patch the MAVProxy installer to do so. I'd also have to figure out a method for users that have built their own MAVProxy though ... I've realised that a `pip install MAVProxy` in Windows will result in the same issue.

It really depends if the user is running MAVProxy from the exe installer or the pip installer. This may require some though...

OXINARF, 2017-02-06:
@stephendade When installing from pip isn't that a matter of Python/pip having itself registered the correct path? I mean, I think that when installing from pip you'll have the same problem for any package, so it should be fixed by the installer (pip in that case).

stephendade, 2017-02-07:
After some research, Python's scripts directory also adds itself to the system path (where mavproxy.py is located after a pip install). Combined with the MAVProxy installer being on the system path, that means:
either mavproxy.exe or mavproxy.py will be on the system path.

So, all sim_vehicle.py needs to do is check the system path for mavproxy.py and mavproxy.exe. If one is found, get it's location and use it for running mavproxy.

OXINARF, 2017-02-07:
That's great! Can you let us know when the MAVProxy Windows installer has been fixed? Then we can fix sim_vehicle (and if anyone complains we can tell them to update their MAVProxy installation).

peterbarker, 2017-02-08:
On Tue, 7 Feb 2017, Stephen wrote:

> So, all sim_vehicle.py needs to do is check the system path for mavproxy.py
> and mavproxy.exe. If one is found, get it's location and use it for running
> mavproxy.

That's a little unclear.  Can we just run "mavproxy.exe" bare of any 
pathing once the Windows installer for MAVProxy has been run?

stephendade, 2017-02-08:
> That's a little unclear. Can we just run "mavproxy.exe" bare of any pathing once the Windows installer for MAVProxy has been run?

There's two methods of installing MAVProxy in Windows:

- Use the installer, which gives you a mavproxy.exe
- Use `pip install mavproxy`, which gives you a mavproxy.py

sim_vehicle.py needs to be able to cope with either install method.

peterbarker, 2017-02-08:
On Tue, 7 Feb 2017, Stephen wrote:

> There's two methods of installing MAVProxy in Windows:
>
>  *  Use the installer, which gives you a mavproxy.exe
>  *  Use pip install mavproxy, which gives you a mavproxy.py
> 
> sim_vehicle.py needs to be able to cope with either install method.

No worries.  Let me know when there's a new Windows installer for me to 
test?

Ta,
-- 
Peter Barker                          |   Programmer,Sysadmin,Geek.
pbarker@barker.dropbear.id.au	      |   You need a bigger hammer.
:: It's a hack! Expect underscores! - Nigel Williams

stephendade, 2017-02-09:
MAVProxy source has been edited. Need to wait for @tridge to do an upversion of MAVProxy, then I can build and upload the new installer.

tatsuy, 2017-02-12:
I think that sim_vehicle.py on Cygwin should search mavproxy.exe using PATH.

My friend runs Windows OS, which is 32-bit (x86).
There is only Program Files folder. (Not Program Files (x86))
He cannot run sim_vehicle.py on Cygwin terminal because mavproxy.exe was not found. 

So he adds the Program Files folder to $PATH.
  echo 'export PATH=$PATH:”/cygdrive/c/Program Files/MAVProxy”' >> ~/.bashrc
  source ~/.bashrc

However, the same problem occurs.
I think this root cause is following code:

https://github.com/ArduPilot/ardupilot/blob/master/Tools/autotest/sim_vehicle.py#L654
if under_cygwin():
    cmd.append("/usr/bin/cygstart")
    cmd.append("-w")
    cmd.append("/cygdrive/c/Program Files (x86)/MAVProxy/mavproxy.exe")

DavidIngraham, 2018-12-01:
Fixed in #9877

closed 2018-12-01 by DavidIngraham