ArduPilot/ardupilot #5669: ./waf copter error

Hyacinthchou, 2017-02-06:
#### Issue details
Missing configuration file '/home/fcs/work/ewt_v1_20170113/build/sitl/ap_config.h', reconfigure the project!

#### Version
master

#### Platform
[  ] All
[  ] AntennaTracker
[ * ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
quad

#### Hardware type
pixhawk

#### Logs
fcs@fcs-virtual-machine:~/work/code/ewt_v1_20170113$ ./waf configure --board px4-v2
Setting top to                           : /home/fcs/work/code/ewt_v1_20170113 
Setting out to                           : /home/fcs/work/code/ewt_v1_20170113/build 
Autoconfiguration                        : enabled 
Setting board to                         : px4-v2 
Checking for program 'arm-none-eabi-ar'  : /usr/bin/arm-none-eabi-ar 
Using toolchain                          : arm-none-eabi 
Checking for 'g++' (C++ compiler)        : /usr/bin/arm-none-eabi-g++ 
Checking for 'gcc' (C compiler)          : /usr/bin/arm-none-eabi-gcc 
Checking for HAVE_CMATH_ISFINITE         : no 
Checking for HAVE_CMATH_ISINF            : no 
Checking for HAVE_CMATH_ISNAN            : no 
Checking for NEED_CMATH_ISFINITE_STD_NAMESPACE : no 
Checking for NEED_CMATH_ISINF_STD_NAMESPACE    : no 
Checking for NEED_CMATH_ISNAN_STD_NAMESPACE    : no 
Checking for header endian.h                   : not found 
Checking for header byteswap.h                 : not found 
Checking for program 'cmake'                   : /usr/bin/cmake 
Checking cmake version                         : 3.5.1 
Checking for program 'ninja, ninja-build'      : not found 
Checking for program 'make'                    : /usr/bin/make 
Checking for program 'python'                  : /usr/bin/python 
Checking for python version >= 2.7.0           : 2.7.12 
Source is git repository                       : yes 
Update submodules                              : yes 
Checking for program 'git'                     : /usr/bin/git 
Gtest                                          : PX4 boards currently don't support compiling gtest 
Checking for program 'arm-none-eabi-size'      : /usr/bin/arm-none-eabi-size 
Benchmarks                                     : disabled 
Unit tests                                     : disabled 
'configure' finished successfully (1.761s)
fcs@fcs-virtual-machine:~/work/code/ewt_v1_20170113$ ./waf copter
Missing configuration file '/home/fcs/work/ewt_v1_20170113/build/sitl/ap_config.h', reconfigure the project!

OXINARF, 2017-02-06:
I'm not sure how you got there, but ap_config.h should be generated in the configure step. Try to do a `./waf distclean`, and then proceed to do the usual configure and build steps.

OXINARF, 2017-07-31:
No answer after several months, closing.

closed 2017-07-31 by OXINARF

lukedempsey, 2018-11-21:
I encountered the same issue. @OXINARF's suggestion fixed it for me, thanks.

Occy88, 2020-07-03:
Thanks for the solution.

KrishnanS2006, 2021-11-21:
I did @OXINARF's solution, and it gave me the same error. Process:
```
./waf distclean
./waf configure
./waf configure --board PixHawk1
./waf plane
docker build . -t ardupilot
docker run --rm -it -v `pwd`:/ardupilot ardupilot:latest bash

cd ArduPlane/
sim_vehicle.py -w
```
Error:
```
Missing configuration file '/path/to/ardupilot/build/Pixhawk1/ap_config.h', reconfigure the project!
```
Using:
Board: PixHawk1
Target: Plane
OS: Ubuntu 20.04

rmackay9, 2021-11-22:
@KrishnanS2006, maybe try removing the "build" directory completely.

KrishnanS2006, 2021-11-22:
> @KrishnanS2006, maybe try removing the "build" directory completely.

Wow, that worked! Thanks!