ArduPilot/ardupilot #5672: Plane: FBWB, Cruise, Loiter modes should be referenced AMSL

WickedShell, 2017-02-06:
#### Issue details
FBWB, Cruise, Loiter flight modes are all subject to changes in altitude if the home position is changed. These are all expected to maintain their current altitude rather then suddenly diving. (If anyone disagrees then you should post so here, but the current thought is these should all switch to useing AMSL altitudes).

#### Version
Master

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover

peterbarker, 2024-07-31:
Still a problem in 2024.  Added a disabled autotest: https://github.com/ArduPilot/ardupilot/pull/27721

Georacer, 2024-11-27:
This can now be closed. This is fixed twice, by https://github.com/ArduPilot/ardupilot/commit/2e2c7b50ad2a6fb4fde9c27d9511ddfc862db794 and https://github.com/ArduPilot/ardupilot/pull/28675

@peterbarker please reopen if you disagree.

closed 2024-11-27 by Georacer