ArduPilot/ardupilot #5675: landing gear will not deploy in landing mode.

geofrancis, 2017-02-06:
#### Issue details
i have landed my hex on its gimbal twice now because the gear would not deploy automatically, i have tested this with landing mode, RTL, and low battery landing mode and it will not lower the gear on its own.

#### Version
3.5rc1

#### Platform
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_

#### Logs
_Please provide a link to any relevant logs that show the issue_

OXINARF, 2017-02-06:
Please provide a log file. Also, this list is for confirmed bugs and feature requests, next time please post first in the forum.

geofrancis, 2017-02-07:
ok i think im getting somewhere with this, i was reading back through something about auto retracts and seen this https://github.com/ArduPilot/ardupilot/issues/5567

lvale: The Auto/Middle is already working and deploys the LGR automatically on the last Land stage (which also includes RTL)

this is different from the documentation here 
http://ardupilot.org/copter/docs/landing-gear.html

If assigned to a 3-position switch, the middle position will not cause a change in state of the landing gear. In the future, this will be used for an automatic mode, but this is not enabled for AC3.3.

does this mean it will not automatically deploy on landing if its not in the middle position?

OXINARF, 2017-02-07:
Yes, you need to have a switch assigned to landing gear and put it in the middle position.

geofrancis, 2017-02-07:
so how do you retract the gear ?

if you retract the gear and dont put the switch back in the middle it wont redeploy on landing? thats a very different behavour from any other version and is not documented. 

gear should ALWAYS deploy on landing, im ok with you being able to over ride it and put them back up again but i nearly lost a gimbal when it belly landed after it hit battery failsafe.

how would this behavoiur work if there was a RC receiver failure?

hackra76, 2017-02-07:
I too had a belly (gimbal) landing after update to rc3.5rc1... I used a 2 way switch without problems with previous versions... Now it behaves very differently.

rmackay9, 2017-05-23:
I think this should all be resolved with AC3.5-rc5.  I've done some tests in the simulator at least and it seems to work.  I'm going to close this issue but if you see issue report them here or in the AC3.5 discussion area.
http://discuss.ardupilot.org/c/arducopter/copter-3-5

closed 2017-05-23 by rmackay9