ArduPilot/ardupilot #5676: check parameter IDs are globally unique in SITL

tridge, 2017-02-06:
we should have an extra check in SITL that parameter IDs in eeprom are unique, to cope with the zero index shifting causing possible collisions

WickedShell, 2017-02-06:
I hit a similar problem to this the hard way the other day, I had duplicate
parameter names which I didn't realize for quite a while, it would be
really nice to check for this in SITL as well (to allow for yelling at
silly devs)

arshPratap, 2020-12-15:
Is this issue still open ?? Because there have been changes similar to what @peterbarker suggested in the AP_Param::check_var_info function

peterbarker, 2020-12-15:
@arshPratap I believe this issue is still open.  I would expect there to be a distinctive chunk of code within a `#if CONFIG_HAL_BOARD == HAL_BOARD_SITL` block.  It would create an array of ~2000 parameter IDs and then walk the parameter tree, generating the storage ID for each parameter and adding it to the list.  If it already exists in the list - that's an error.

It would look something like `count_parameters`.

I think `get_key` is the relevant call to find the ID to check.

Testing would include deliberately creating the scenario outlined in the issue description.

Again - not for the faint of heart.  It's hairy code.

arshPratap, 2020-12-15:
Thank you sir for the information..I will keep this in mind while working on the issue.

arshPratap, 2020-12-15:
Also, I had a doubt, I tried using hashmap to improve the N^2 algorithm(duplicate key function). Is using hashmap a good idea for embedded driver development considering the memory constraints??
Thanks a lot in advance!!

peterbarker, 2025-06-05:
Turns out that @WickedShell actually gave us this check in July 2017.  Thanks @WickedShell and apologies to @arshPratap for having him chase a solved problem!

peterbarker, 2025-06-05:
> Also, I had a doubt, I tried using hashmap to improve the N^2 algorithm(duplicate key function). Is using hashmap a good idea for embedded driver development considering the memory constraints?? Thanks a lot in advance!!

Solution used a bitmask to track which values where used.

closed 2025-06-05 by peterbarker