ArduPilot/ardupilot #5679: barometer alt invalid on commit ac90527d

budebulai, 2017-02-07:
#### Issue details
I downloaded ArduPilot with master version commit ac90527d
This version can't measure altitude from barometer.
ArduPlane has not this problem.
What's wrong with this version?

#### Version
commit ac90527d

#### Platform
[  ] Copter

#### Hardware type
pixhawk

budebulai, 2017-02-07:
https://github.com/ArduPilot/ardupilot/issues/4225

OXINARF, 2017-02-07:
That's a very old version from master. Please update to a current master, then if there's an issue report back. For now, I'm closing this.

closed 2017-02-07 by OXINARF

budebulai, 2017-02-07:
At first I didn't know this bug, so I develop on this version, using rtk and gps heading, adding some code to ekf2, and everything is ok when gps is valid. 
One day, quadcopter fell down when gps lost abruptly, and I realised baro altitude was not useful.
I guess there is something wrong with ekf2

来自 魅族 MX5

时间：周二 2月7日 22:02
收件人：ArduPilot/ardupilot <ardupilot@noreply.github.com>
抄送：budebulai <zhaolun26@163.com>,Author <author@noreply.github.com>

OXINARF, 2017-02-07:
If you don't want to worry about bugs like that you shouldn't develop on top of master but on top of a stable branch, like Copter-3.4.

Master changes fast and, although we try not to introduce bugs, it can happen. We appreciate bug reports about issues in master, but not from a version that is more than 6 months old. There was 3065 commits, 1687 changed files (with 80513 additions and 45751 deletions) after that...