ArduPilot/ardupilot #5684: Prevent code duplication between guided/auto/manual modes

ClementChristomanos, 2017-02-07:
There is currently a lot of code duplication between some parts of the current architecture : 

- Guided mode is almost identical to auto mode code
- There are very slight differences between standard land code, auto land code and RTL land code
- Pretty sure to find some other examples of code duplication.

I start this Issue so we can discuss what could be done to prevent code duplication. This would allow to safe some flash space, but could also prevent potential crashes. 

I crashed costly hardware yesterday because of a bug, that was fixed in Auto mode, but the fix wasn't done in Guided mode, on the same function. Of course, I'm not blaming anyone for that, but I think something can be done on the architecture to prevent such things to happen in the future, when future bugs appear. 

This may become a big refacto for the current architecture, let's discuss it :-)

rmackay9, 2017-02-07:
Yes, there tend to be subtle differences between the various forms of land, takeoff, etc.  There's been some work on land mode but certainly there's more that can be done to reduce duplication across the flight modes.

By the way, @peterbarker has done some work to transform the flight modes into objects for both copter and rover.  There is a PR marked for AC3.6. Perhaps that's part of the solution.

ClementChristomanos, 2017-02-07:
Thank you for that info Randy, I'll check the PR from @peterbarker !

OXINARF, 2017-02-07:
I think everyone can agree to this. Like @rmackay9 said, PR #3802 is the first key to this (and other improvements). I hope we can merge it for 3.6.

peterbarker, 2017-02-08:
Yes!  Factoring is definitely on the cards after the flight-modes patch.  I'm leery of doing any more work along those lies myself ATM because I don't want to conflict with the flight-modes patch.

There's also https://github.com/ArduPilot/ardupilot/pull/5227

amilcarlucas, 2017-08-23:
ping

peterbarker, 2017-12-19:
Is there any clear milestone which could be reached that would close this issue?

I'm not sure our issues list is a great place to "discuss" this ;-)

amilcarlucas, 2021-03-09:
The two PRs discussed above have been merged, and @peterbarker is constantly removing duplicated code. So closing this

closed 2021-03-09 by amilcarlucas