ArduPilot/ardupilot #5686: Septentrio GPS only works on GPS1

edwinhayes, 2017-02-07:
#### Issue details
The Septentrio GPS (SBP) driver only seems to work if the GPS is connected to GPS port 1; if connected to GPS2, it sits attempting to connect (think it's autobauding?) forever.  Swapping the same receiver to GPS port 1 fixes the issue, and GPS2 still works fine with other receivers (tested w. std uBlox).  There isn't really any obvious reason for this limitation.

#### Version
AC3.4.4

#### Platform
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover

#### Airframe type
Irrelevant.

#### Hardware type
Pixhawk v2.1 Beta

#### Logs
No relevant logs.

meee1, 2017-02-07:
did you set the gps_type2?

edwinhayes, 2017-02-09:
(Sorry, took a while to get one of the guys to redo the test.)

Yes, GPS_TYPE2 is set appropriately.  Power cycling before/after setting the parameter doesn't seem to change anything.  Having GPS_TYPE set to uBlox, Septentrio, or auto at the same time makes no difference; neither does having auto config turned on or off.

Having looked at the code, I have no idea why this doesn't work; no obvious special treatment of the first GPS port.

OXINARF, 2017-02-09:
Just to be clear:

- you have a working GPS connected to GPS1 port
- you have a Septentrio GPS connected to GPS2 port not working
- you have SERIAL4_PROTOCOL set to GPS and GPS_TYPE2 set to AUTO or SBP

Is that all correct?

edwinhayes, 2017-02-09:
Yes, that sounds correct: 

uBlox on GPS1 + Septentrio on GPS2 = uBlox works, Septentrio doesn't.
Septentrio on GPS1 + uBlox on GPS2 = Both work.

Yes, serial protocol set for GPS, and GPS type set for SBF (I believe SBP is used to denote the Swift Nav binary protocol?).  Is using AUTO for GPS type expected to work with Septentrio units (which require configuration)?

OXINARF, 2017-02-09:
Ah, sorry, I was fouled by your original post saying SBP :smiley: Yes, you are correct that it is SBF and you indeed need to set it explicitly.
So, with it explicitly set, how long have you waited to get it detected? Do you receive any MAVLink message announcing a GPS was detected?

edwinhayes, 2017-02-09:
Yup, the status text message that's something like "GPS2 detected as SBF" (without a baudrate) repeats over and over; which I think indicate's it's trying to autobaud the connection?

OXINARF, 2017-02-09:
Yes, it should be trying to find the right baudrate. I've looked at the code and I can't find anything that makes it different. Since I don't have a Septentrio GPS to test, I'll have to let other people try to figure it out.

WickedShell, 2017-02-10:
> Assigned #5686 <https://github.com/ArduPilot/ardupilot/issues/5686> to
> @WickedShell <https://github.com/WickedShell>.

I'll get out my hardware over the weekend for testing. I will only be able
to test on a Pixhawk however due to cabling issues...

jmachuca77, 2017-02-10:
Its working on my setup on GPS2 with no problems.... but I have the baud rate set explicitly. Not Auto baud rate.

edwinhayes, 2017-02-21:
Hmm, ok.  I guess I'll have to investigate: the staff member who reported this to me is away at present, so will follow up with them on return.

@jmachuca77 - Would you be able to tell me which hardware and software versions you're running in your working setup?  And possibly dump the relevant parts of your parameter table for me to compare against?

WickedShell, 2017-07-09:
Closing as a support issue. Just validated that GPS1/2 both work no problems on a commit near master.

closed 2017-07-09 by WickedShell