ArduPilot/ardupilot #5687: CAM & TRIG messages inconsistent

edwinhayes, 2017-02-07:
#### Issue details
When using a camera without a hot-shoe trigger input to the flight controller, CAM messages are logged to dataflash.  When using a camera with a hot-shoe input configured, both CAM and TRIG messages are generated.  This would seem to indicate then that the CAM messages were spawned at the time of sending the camera shutter command, with the TRIG message being spawned by the hot-shoe input event - however, it's actually the opposite way around: the TRIG message indicates the shuttering triggering command, so it's slightly weird that you get CAM messages at all if you don't have an input set up.  For consistency, it would seem more sensible if both CAM and TRIG dataflash entries were generated together if the hot-shoe input is not configured, so that analysis can be handled identically regardless of which type of configuration is used.

#### Version
AC3.4.4

#### Platform
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover

#### Airframe type
Irrelevant.

#### Hardware type
Pixhawk V2.1

#### Logs
Probably not required.

WickedShell, 2017-02-07:
This was by design as it allowed all existing photo tagging software to
continue to work without any changes, I agree its inconsistent, however
because of that history I don't think it makes sense to change it now.

edwinhayes, 2017-02-07:
Would spawning both messages regardless break any existing software?  It doesn't seem immediately obvious to me that it would break anything, assuming both messages denote the same time/location.

WickedShell, 2017-02-08:
Spawning both is probably fine, at least the tagging work flows I know of off the top of my head would work with this without any problem.

OXINARF, 2017-02-08:
I don't know much about this, but when (and for what) would you want to use TRIG messages?

edwinhayes, 2017-02-08:
Not for the actual tagging I suppose, but can use the TRIG messages for diagnostic purposes, since then all is working correctly, you expect a 1:1 relationship between TRIG and CAM messages?

WickedShell, 2017-02-08:
You expect it, but the current best you can do is almost always N trigs
with N+1 CAM's (for some reason you get a false positive on the first
feedback event). But this only applies to the situation where feedback is
enabled, otherwise you don't have any valid metric without it anyways.

edwinhayes, 2017-02-08:
Ah, ok: I think currently on our aircraft it's something like the opposite, because we use the first few shutter signals to queue other payload functions anyway.  So long as it's some known difference, still useful for diagnostics.

Related, I guess the other reason for using TRIG messages in the presence of the hot-shoe input would be that if there was something wrong with the input so you found you had no CAM messages, you'd still be able to tag using the TRIG messages in order to somewhat salvage the dataset (though whether it would still be fit for purpose is another issue entirely).  Also not relevant in the case you don't have a trigger input.

So, with respect to this issue: I guess my position is that it would be a slight improvement in terms of consistency to generate both message types, regardless of whether a trigger input is used, but probably the only feasible way of determining whether it breaks anyone's tagging would be to try and see if anyone objects, since it's probably only weird workflows which would be affected.

OXINARF, 2017-02-08:
My thinking about this is that the CAM message gives you the best possible information of when your camera has taken the shot. Looking at both cases:

- with a hot-shoe you get a CAM message when the camera actually shoots and a TRIG message for when ArduPilot sends the signal to the camera
- without a hot-shoe you get a CAM when ArduPilot sends the signal to the camera

So my question is: what do you expect the TRIG message to be in the second case? The same as a CAM message? If yes, then what is the purpose of it? My questions about this are because we already write a lot of information into the log, so writing "repeated" messages isn't desirable.

edwinhayes, 2017-02-09:
If the TRIG message doesn't convey useful information, I don't have a problem getting rid of it entirely (except, as previously mentioned, it might be required for backwards compatibility).

But I'm strongly in favor of AC being as consistent as possible, so if there are both messages under some circumstances, I'd prefer there always be both.  I would have thought this consistency was more valuable than the overhead of logging the additional message, given that camera events are pretty infrequent compared to other log messages.

WickedShell, 2017-02-09:
I'd say that in some manner it is, CAM is your best guess for when the
camera took a photo period, that is always true regardless of mode. Now we
can always emit the TRIG messages, it's not a ton of data to add to the
log, but I can see either side of the argument and am willing to go either
way. I just want to caution that I don't see adding it as enhancing any
proposed use case yet, which lowers the enthusiasm to making a change.

edwinhayes, 2017-02-09:
I've no problem with this being closed "won't fix", or any alteration being deferred until there is some other substantive change to the camera logging functionality (#5688 perhaps).  Mostly just wanted to know whether there was some deliberate thinking behind the current behaviour.

OXINARF, 2017-02-09:
> If the TRIG message doesn't convey useful information

I think it can have useful information in the case you use a hot-shoe, but in the case you don't it would be a duplicate message (for which I've seen no value).

> But I'm strongly in favor of AC being as consistent as possible

The consistency would be achieved with always logging a TRIG message and a CAM only with a hot-shoe. Issuing the CAM message at different times is really what is inconsistent. As @WickedShell explained, it was done that way to keep backwards compatibility which was deemed more important.

> I would have thought this consistency was more valuable than the overhead of logging the additional message, given that camera events are pretty infrequent compared to other log messages.

Well, the issue with that thinking is that here and there people will say the same, "it's just a small message from time to time, it doesn't hurt", but all together it will then matter.

I think that adding the TRIG message in all cases is only worth it if, in the future, we are going to remove the CAM message when there's no hot-shoe. This would give time for other software to change their parsing so that they look first for CAM messages and if those don't exist then use TRIG messages - unfortunately I believe this wouldn't happen and they would only change once we made the complete change (with the potential to affect our users).

CraigElder, 2017-02-09:
The TRIG message was added to better synchronize the position and attitude data recorded in the DF log with the actual time that the photo was taken.
The thinking with regard to backwards compatibility is that Pix4D only the CAM message so it should be retained until we can get them to update their parser. At the moment when I use the TRIG message I do some massaging with the DF log to remove the CAM message and then replace the TRIG delimiter with CAM.
Unfortunately our efforts to get Pix4D to update their parser have not been very successful so this is the current work around.

WickedShell, 2017-02-10:
@CraigElder Uh what? You want to use the CAM message as thats the feedback location from the camera rather then the time when we told it to take a photo. The only reason to not use the CAM is if your feedback solution has some delay constant, in which case you already don't have a very good solution. You should be tagging with the CAM though.

edwinhayes, 2017-02-10:
Having now looked at the code, it seems like the CAM and TRIG messages are basically identical anyway (differing only in what I assume is the message ID field - but both end up calling `Log_Write_CameraInfo`)?

This seems fairly redundant - it seems clear to me that there should be exactly one message which contains location information; and one message which only contains information about the time the shutter was actuated at, to allow diagnostics.

So if logging overhead is a legitimate consideration, I'd have thought the most obvious thing to do is stop duplicating information between the TRIG and CAM messages.

WickedShell, 2017-02-10:
The duplication isn't there at the moment. By the time the feedback arrives
you are at a new location/attitude so you need to log that information
instead. You in no way are interested in the TRIG message unless something
malfunctioned with your feedback mechanism, or if you are trying to assess
if you got all the photos..

edwinhayes, 2017-02-10:
That's basically what I meant: there is no need for the TRIG message to have position information if it's primary purpose is to indicate the shutter actuation occurred for diagnostic reasons - you don't care where that happened, just whether or not it took place.

Although, if TRIG didn't have location, you wouldn't be able to use it to fall back to if your CAM messages weren't there, though I don't know if this is actually a useful function.

WickedShell, 2017-02-10:
Fallback is a valid and useful feature (that I've actually had to use), so
I'd be opposed to deduplication of the messages for that reason :)

jmachuca77, 2017-02-10:
There is a fundamental misunderstanding here. The CAM messages are always the messages that should be used for Geotagging. 

In the case when there is no hotshoe feedback, the CAM messages are written when we send the shutter command because there is no other way to get feedback of when the actual pictures is taken. Here no TRIG command is necessary because it would be at the same exact moment the CAM messages where taken.

In the case when there is a hotshoe, then the CAM messages contain the precise time, when the feedback is received from the camera, making the Geotag much more accurate. In this case the TRIG message is used to log when we send the trigger command. 

The TRIG message was added so that we could see what the delay was between sending the shutter command and actually taking a picture. 

This is consistent, because the CAM messages are always meant to have the most accurate location possible for the geotag.

WickedShell, 2017-02-10:
Agreed. Closing as non issue.

closed 2017-02-10 by WickedShell