ArduPilot/ardupilot #5688: Add parameter to select CAM data source.

edwinhayes, 2017-02-07:
#### Issue details
According to log file documentation, CAM dataflash messages are populated with aircraft position based on accelerometer plus GPS information (unclear if this just means the fused state output from the EKF or something else).  This is probably fine for standard GPS implementations, but isn't adequate for aircraft running RTK GPS units (or other localisation).

There should probably be a parameter which allows selection of what data source is used to populate CAM messages; allowing selection between fused state, any single GPS receiver, or [any hypothetical other localisation source].

#### Version
AC3.4.4

#### Platform
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover

#### Airframe type
Irrelevant.

#### Hardware type
Pixhawk 2.1

#### Logs
Probably not required.

CraigElder, 2017-02-09:
>>>There should probably be a parameter which allows selection of what data source is used to populate CAM messages;

The CAM and TRIGG messages already have both the the GPS and EKF position and attitude information in them and are specifically configured to use with RTK GPS.

That means you get all of the best available position and attitude data and you can select the fields you want to use during post processing.

https://github.com/ArduPilot/ardupilot/blob/master/ArduCopter/commands_logic.cpp#L1162
 DataFlash.Log_Write_Camera(ahrs, gps, current_loc);

https://github.com/ArduPilot/ardupilot/blob/master/ArduCopter/commands_logic.cpp#L1166
DataFlash.Log_Write_Trigger(ahrs, gps, current_loc);

I find this workflow works very well. I would not want to chose to log only one set of data prior to a flight.

edwinhayes, 2017-02-09:
Documentation for the CAM message (http://ardupilot.org/copter/docs/common-downloading-and-analyzing-data-logs-in-mission-planner.html) indicates that logged altitude is based on "The accelerometer + barometer estimated altitude in cm above ground", and does not indicate that EKF and GPS information is logged separately.  Is the documentation incorrect?

edwinhayes, 2017-02-09:
Oh, inspection of the underlying code seems to answer that question.  Related then: is there some canonical location documenting the dataflash log format, other than the code itself?

Regardless, it looks like there is GPS altitude recorded separately, but only EKF fused horizontal position?

WickedShell, 2017-02-10:
@CraigElder we don't log GPS position, @edwinhayes is correct on that https://github.com/ArduPilot/ardupilot/blob/master/libraries/DataFlash/LogFile.cpp#L1781 You could correlate the GPS timestamp to GPS messages, but you are vulnerable to skips with that approach...

edwinhayes, 2017-02-21:
Ok, some further information: based on comparison between photos tagged directly by our RTK GPS receiver and the same images tagged by Mission Planner, we see typical horizontal discrepancies of (RMS distance) 1.1m in the direction of flight, and 0.35m orthogonal.  This makes the CAM messages unsuitable for use in RTK driven photogrammetry.

(There are also large vertical differences, but since GPS alt is included in the CAM message, I'll assume at this point that it's just Mission Planner not using it)

Presumably, this is because the EKF doesn't trust GPS position sufficiently.  Not sure what the current behaviour is, but the EKF should be adjusting its weights based on the type of lock reported, and be almost entirely favouring (interpolated) GPS when RTK Fix mode is reported?

Failing that, I circle back to my original proposition: either CAM messages should include both fused and (interpolated) GPS data, or there should be the option of selecting the source to use via parameter.

CraigElder, 2017-02-22:
Are you using the Septentrio GPS to tag the photos?

edwinhayes, 2017-02-22:
Yes, that's correct.

edwinhayes, 2017-02-22:
Fundamentally, we'd be happy to use the GPS for tagging, except that we'd like to be able to perform lever-arm correction using the aircraft's attitude to compensate for the location of the antenna with regards to the camera - and since the GPS doesn't know the attitude of the aircraft, it won't perform this correction by itself.

meee1, 2017-02-22:
could I get a copy of your data set? just so I can see what exactly is going on?

CraigElder, 2017-02-22:
@meee1 Last night I sent a message to @edwinhayes @Aeronavics so we can sort out what is going on

edwinhayes, 2017-02-22:
@meee1 - Not sure what we agreed with the owner of the site, might have to sanitise some data first?  Will investigate.  Otherwise might just need to get someone to fly another mission.

@CraigElder - Sorry, not sure specifically which message you're referring to?  I possibly don't know how to drive this thing properly.

proficnc, 2017-02-22:
Would I not be correct in assuming this is directly related to the .7 rather than .9 decimal places, truncating the position as we discussed at the Australian meet up?

So this would be a GPS driver / EKF / Copter mod required, as well as uploading the data from MP into Ardupilot

WickedShell, 2017-02-22:
@proficnc you would be incorrect, 7 decimal places at the equator is good to ~11mm.

WickedShell, 2017-02-22:
@edwinhayes you can check how accurate the most recent GPS tags are that are above/below the CAM tag you used to tag a photo. Even if you just pull that by hand it would be an easy way to validate if that would solve the problem you are seeing.

edwinhayes, 2017-02-22:
@WickedShell hmm, yeah that's probably a good idea thanks.  I'll get someone to have a go and report back.

proficnc, 2017-02-22:
Aerobatics is not at the equator... they are in NZ... loosing them 2 decimal places in the coordinates

proficnc, 2017-02-22:
Autocorrect, Aeronavics

edwinhayes, 2017-02-22:
@proficnc - This probably reaches the limit of my GPS understanding: why does our location in NZ (although to be pedantic, the data set we've been using for analysis was from Western Australia) reduce the ground accuracy?  I'd have thought the decreasing circumference at higher latitudes reduced the ground distance each degree of longitude was spread over?

CraigElder, 2017-02-22:
@edwinhayes I sent a message to contact @ aeronavics.com

mboland, 2017-02-22:
As the 7 or 9 decimal place has surfaced again, and a surveying consultant I work with lurks here, he has hit me with this which I am sharing to broaden the knowledge of those here.
I hope this is OK and not just unwanted noise.
Mike
 
I’m not sure about the equator but here in Australia 1 second is about 30 metres depending on where you are (could be 25 could be 34 could be 28 – also differs in distance between north-south and east-west from location to location)
 
In Central Queensland we use MGA94 projection which is a Transverse Mercator (TM) projection of the (Geocentric) GDA94 datum (the geographic one). MGA94 zones are typically represented in metres. GDA94 uses the same spheroid as WGS84 and is ITRF compliant.
 
So using a seven parameter transformation between geographic coordinates (GDA94) to TM coordinates (MGA94 zone 56) on actual coordinates we get the following
Seven Decimal Places: 151.2943275E, -23.8655136S = 326317.031E, 7359616.631N
 
    Above  +0.000 000 1: 151.2943276E, -23.8655137S = 326317.083E, 7359616.631N
 
As you can see there is little difference in the Northing but about a  5cm diff in the easting when you use 7 decimal places and increment by 0.000 000 1. the differences obtained by 1/10 000 000 varies in the easting and northing values depending on where you are in Australia as longitudes converge at the poles and latitudes run parallel.
 
Also I’m not sure how the pixhawk/missionplanner handles decimal places (rounding or truncation) plus what is the precision of the EKF filters and does it change the value of the last decimal place (or last two decimal places.) should EKF be operating at 8 or 9 decimal places?
 
Fortunately, and that’s if the Pixhawk can actually talk to the Septentrio, we won’t be using the coordinates that the pixhawk/missionplanner stores as we will be logging direct to Septentrio using the coordinates of the survey grade L1/L2 GNSS module. That’s if the bloody Pixhawk will behave!
 
 
Hope this helps
 
Jason

CraigElder, 2017-02-23:
@edwinhayes What value do you have for the GPS_DELAY? http://ardupilot.org/copter/docs/parameters.html#gps-delay-ms-gps-delay-in-milliseconds

CraigElder, 2017-02-23:
I just found a message from Septentrio that they calculated the delay to be 100ms

CraigElder, 2017-02-23:
We need to run https://github.com/ArduPilot/pymavlink/blob/master/tools/mavgpslag.py on the dataset