ArduPilot/ardupilot #5693: Copter: merge Solo and Maxell smart battery drivers

rmackay9, 2017-02-09:
In the AP_BattMonitor directory we have two smart battery drivers, they are:

- Solo uses internal I2C bus, PEC (error correction) and reads manufacturer specific values for current, battery capacity and "is button pushed"
- Maxell uses the external I2C bus and simply reads voltage and current using standard registers

We should be able to merge these two if we can:

- tell them apart which should be possible using the manufacturer name register
- use the SpecificationInfo command and checking the revision number to see if PEC is used

geofrancis, 2017-02-09:
specifically what maxell sensor will be supported? would it now be possible to do a diy smart battery and would this work with multiple batteries?

rmackay9, 2017-02-09:
@geofrancis, I'll be writing up a wiki page once I've collected all the info on these Maxell batteries.  I suspect the new Maxell driver will actually work with many smart batteries.
We support two battery monitors but the problem is conflicting SMBus/I2C addresses.  So unless the battery itself can be configured to use a non-standard address only one could be attached.  The other alternative is to use the internal I2C bus but that's not exposed on most flight controllers.

rmackay9, 2017-03-29:
After thinking about this more, at this stage I don't think we should try and merge the Maxell and Solo smart battery drivers.  They're simply too different.  When other smart batteries become common and we write drivers for them, we should try and extend either the Solo or the Maxell driver instead of adding yet another driver.

closed 2017-03-29 by rmackay9