ArduPilot/ardupilot #5695: Clean way to add a vehicle

Dermiste, 2017-02-09:
Hi. I am going to fork the repository in order to add a new vehicle directory, for a bi motor copter (Avatar type). What would be the correct way to do that in order to fit well with the waf build interface?

For now I have copied the ArduCopter directory and renamed it ArduBimotor as well was the main cpp file inside it. I also changed in the Makefile for that folder the name of the command called in the upper Makefile for a new "bimotor" replacing the old "copter". What would be next? Any help would be much appreciated (at least a list of file to modify). 

Cheers development team !

Thomas

khancyr, 2017-02-09:
Hi,
It is better to ask support on discuss forum .
About your question, Arducopter is generic enough to accept multiple copter type. I am not specialist on copter, but I would look in https://github.com/ArduPilot/ardupilot/tree/master/libraries/AP_Motors to see how different frame are managed currently !

Dermiste, 2017-02-09:
Thanks ! Yes it sounds actually better to add another configuration dedicated to a Bicopter instead of forking the entering thing. I'll look into that.

OXINARF, 2017-02-09:
Like @khancyr said, what you want is a new frame class, not a new vehicle type. Look at the way things are done for tri, single and coax-copter, those should help you out.

I'm going to close since it's not appropriate for this list. For development questions please post in the forum or ask in Gitter (http://gitter.im/ArduPilot/ardupilot), we'll try to help.

closed 2017-02-09 by OXINARF