ArduPilot/ardupilot #5696: Copter: Simple movements using SET_ATTITUDE_TARGET from Raspberry Pi without GPS

aoigetei, 2017-02-09:
Hi all, 

I am trying to perform simple movements using SET_ATTITUDE_TARGET from a raspberry pi, but my copter does not respond to my commands. I have written the following functions below, and when I attempt to use the "forward" function to pitch forward a specified distance I get no response from the vehicle. 

I have tested out these functions in Gazebo using SITL and it seems to work fine in simulation, but does not work on the actual platform.

This is my platform setup:
- My quadcopter is running Copter: 3.3.3 on a Pixhawk
- No GPS - Using Optical Flow and Rangefinder for altitude control
- Takeoff in GUIDED mode

       def forward(self, distance, speed):
		self.__vehicle.groundspeed = speed
		duration = distance/float(speed)
		t_end = time.time() + duration

		while time.time() < t_end:
			self.__set_copter_attitude(0, 0.1, 0, rotation=1)  -> pitch vehicle until time expires 
		time.sleep(1)

        
       def __set_copter_attitude(self, roll, pitch, yaw_rate, rotation=1):
		if rotation:
			rotate = 1
		else:
			rotate = 0
		msg = self.__vehicle.message_factory.set_attitude_target_encode(
			0,     
			0, 0,   
			0b00000000, 
			[rotate, roll, pitch, 0],  -> attitude quaternion (w, x, y, z | zero-rotation is 1, 0, 0, 0)
			0, #roll rate
			0, #pitch rate
			yaw_rate, 0.5)    # yaw rate, thrust 
		# send command to vehicle
		self.__vehicle.send_mavlink(msg)

Any help is greatly appreciated.
Thank you!

OXINARF, 2017-02-09:
This issues list is for confirmed bugs and feature requests. User support questions should be posted at the forum (http://discuss.ardupilot.org). For this reason, I'm closing this.

I recommend you update to a more recent version, namely 3.4, since the message you are using isn't supported in 3.3.3. You said it worked on SITL, probably you are using master branch there, I recommend you use the same on SITL as you use on your copter.

closed 2017-02-09 by OXINARF

aoigetei, 2017-02-13:
Thank you for the reply, so I have tried upgrading to 3.4. Since I performed the upgrade I am no longer able to arm and takeoff just using the rangefinder and optical flow like I was using version 3.3.3.

 Is there a reason for this ? 

or some new parameter value that is preventing this ?

OXINARF, 2017-02-13:
Post in the forum please, this is not the place for that discussion. If you aren't able to arm look at the error in your GCS.