ArduPilot/ardupilot #5700: Quadcopter sudden descent in Loiter mode.

sashan13, 2017-02-10:
#### Issue details
I've been loitering my 4.5kg quad in LOITER in exactly same position at around 25m for a few minutes. Then it suddenly started descending at around 1.5m/s on itself, then it recovered from descend 2 m above ground. I switched it to althold and landed it immediately after that in that same place.
During descend the althold PID's I - pida.i - parameter stopped changing and flatted out. 

It's worth noting tha ambient temperature was around -5 celsius but the quad has fully enclosed fuselage with vent grates.

#### Version
APM:Copter V3.4.2 (6076bdfc)

#### Platform
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover

#### Airframe type
4.5kg quad
#### Hardware type
Pixhawk+drotekM8N+external HMC5883L
#### Logs

Full log. See at around 7th minute of flight.
https://www.dropbox.com/s/zalo2chkim078ln/2017-02-09%2015-27-46.bin?dl=0

mboland, 2017-02-10:
The correct place to post this would be the user support forum rather than the developers issues list.

The current release version of Copter is 3.4.4 so an update would be recommended.
Do a test flight with more logging turned on (you have IMU and others turned off)
And post your findings at [http://discuss.ardupilot.org/c/arducopter](http://discuss.ardupilot.org/c/arducopter)

You will get a much better response than posting here.

OXINARF, 2017-02-10:
@sashan13 Like @mboland said, this list is for confirmed bugs and feature requests. Please follow the recommendations from Mike and then post in the forum.

closed 2017-02-10 by OXINARF