ArduPilot/ardupilot #5711: COPTER V3.4.4: Can not ARM in Loiter Mode, even when the green light is on

dongfang-a, 2017-02-12:
#### Issue details
I cannot ARM my Quad Copter, even when the green light is ON, in Loiter Mode.

Mission planner told me error message: "GPS2 miss configure. " However, I don't have GPS2. I tried to set  GPS_TYPE2 to disable.  But after that, I cannot let the green light ON in Loiter Mode (it is always blue).

#### Version
COPTER V3.4.4

#### Platform
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover

#### Airframe type
Quad Copter

#### Hardware type
Pixhawk
GPS Port: M8N GPS
Telem1 Port: 3dr Radio
Telem2 Port: Micro mini OSD

#### Logs
Please find my config in attached Logs

[51.zip](https://github.com/ArduPilot/ardupilot/files/769760/51.zip)

rmackay9, 2017-02-13:
It should be displayed on the ground station "HUD" what the reason for the arming failure.  By the way, normally we don't do support in GitHub.  It's better to raise issues in the forums so I'll close this.
http://discuss.ardupilot.org/c/arducopter/copter34

closed 2017-02-13 by rmackay9

WickedShell, 2017-02-14:
No messages about arming failure in the log either.