ArduPilot/ardupilot #5721: Some old uBlox GPS units fail to configure reliably

peterbarker, 2017-02-14:
#### Issue details
Older uBlox GPS units fail to configure correctly in ArduPilot.  Symptoms are these messages on the console:

GPS 0: detected as u-blox at 115200 baud
GPS 0: detected as u-blox at 115200 baud
GPS 0: detected as u-blox at 115200 baud

We currently believe that that message is somewhat misleading; there's a problem in detection where the serial buffer is not cleared before configuration messages are sent, so old data can be interpreted as responses to new queries.

#### Version
master, Copter 3.5rc series

#### Platform
[ * ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover

#### Hardware type
uBlox GPSs
PixRacer
px4-v1

iseries1, 2017-02-24:
It seems that this has been fixed by 5738.

OXINARF, 2017-02-24:
Yes, it was. Thanks @iseries1

closed 2017-02-24 by OXINARF

lthall, 2017-10-09:
I appear to be getting the same issue with Plane 3.8.2 and Copter 3.5.3.

Are there any more fixes?

WickedShell, 2017-10-09:
@lthall are you familiar with how to download all the settings off the GPS? This is usually an indication that you have a much more edge case setup then normal. When @peterbarker hit this it was related to one of the core messages used to keep the GPS driver alive wasn't enabled, and it was to far down the automatic configuration list. If you can get the GPS config then it should be straightforward to figure out why it's failing.

And whats the hardware version?