ArduPilot/ardupilot #5725: feature request: bidirectional frsky telemetry

geofrancis, 2017-02-14:
Now that opentx 2.2 introduces bidirectional telemetry it should be possible to configure arducopter from lua scripts over the frsky rc link.

beta flight have a frsky to Msp bridge working so they can configure vtx  and pids something like that would be very useful for tuning, calibrating accelerometer, compass etc

https://github.com/betaflight/betaflight/issues/1311
#### Platform
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover

floaledm, 2017-02-14:
Hi,

I think that's a great idea. I don't think that there's a bridge capability to the ArduPilot parameters currently and am not familiar with how MSP works exactly, but I'm willing to participate in this. I'd welcome others to participate in this dev as well...

geofrancis, 2017-02-15:
Msp is specific to multi wii derived firmwares like clean flight and beta flight so I don't think it could be copied over I'm just using it as an example of what could be done.

peterbarker, 2021-07-30:
This was done

closed 2021-07-30 by peterbarker