ArduPilot/ardupilot #5732: Building Arducopter with Arduino

thjubeck, 2017-02-15:
Hey all,
I am able to compile the base code with arduino ide for the apm 2.5 but after uploading the code the board does not start (CLI is not accessable and no lights on board). I can upload pre compiled hex and everything works fine.  I have the "special" arduino version installed as well as the tools described in the dev wiki on using the arduino ide. Anyone have any ideas what could be going wrong?

hiro2233, 2017-02-15:
hi @thjubeck, the code work for apm 2.x boards, if you have a special board then you need to modify the code to work correctly.

OXINARF, 2017-02-15:
@thjubeck We don't support APM boards anymore. Code is available but we don't build for them, so knowledge about it is mostly lost. Posting in the forum is fine, but this is a list for confirmed bugs or feature requests for current code, so I'm going to close this.

closed 2017-02-15 by OXINARF