ArduPilot/ardupilot #5743: Feature Request: Prevent Copter execute OLD Mission

dongfang-a, 2017-02-16:
#### Issue details
Last weekend, I used Tower App to plan a mission, but forgot to upload it to the Copter. When the 'auto' mode started, the Copter flied toward the old mission way points (I uploaded it one day ago), which is 10 miles away.

Sooner after 'auto' mode started, the Copter hit a tree and crashed.

I'd suggest add below features:

Add a switch that if true then forgot old mission after reboot.
Or reject entering Auto mode, if the first waypoint (that has lon/lat) is far from home. (need add a parameter)

#### Version
Arducopter 3.4.4

#### Platform
[*] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
N/A

#### Hardware type
pixhawk

#### Logs
N/A

davidbuzz, 2017-03-08:
I think this request is unlikely to be implemented as there are a number of capabilities that rely on missions being "remembered" ... particularly in-flight-reboot which can occur as a result of brownout or electrical connection issues.    However, if you wish to see this feature implemented for your own needs, you can create a PR, or you can contact one of the companies that are offering commercial services with ardupilot.

magicrub, 2017-03-08:
@dongfang-a I'm afraid that even if there was a "switch" (param) to allow this feature, it would certainly be disabled by default and thus you would most likely have it off and to would not have helped you - at least this first time.

OXINARF, 2017-03-23:
@davidbuzz Why close this? We have lot of feature requests that nobody in the team is pursuing - usually we only close it if the idea would be rejected even if the code was contributed. And I can't see why this would be a rejectable one, as long as it's optional.
Also, you talk about in-flight-reboot, but that's not something we support.

WickedShell, 2017-03-23:
Seems quite worthwhile to me, I actually have forced people to do this as part of a planning step (GCS doesn't allow you to get to arming without lots of warnings). However for other GCS's this would be quite valuable. It's similair to the reason we have a RALLY_LIMIT_KM parameter for rally points to make sure you don't fly to old rallys you forgot to clear.

OXINARF, 2017-06-06:
#6383 was opened with basically the same request. I've closed it now, but it has additional comments that might be interesting to look at for someone implementing this.

ChrisBird, 2017-08-28:
Note as its a library change it will impact all platforms that use the AP_mission lib.

commit 46c982d6 (2017-08-29): AP_Mission: add OPTIONS parameter

First option added now is to clear the mission on reboot
Fixes #5743
closed 2017-08-29 by OXINARF