ArduPilot/ardupilot #5745: Race in init order sometimes cause HardFault

night-ghost, 2017-02-17:
screenshot of call stack
[image]

to fix, just alter GCS_MAVLink/GCS_MAVLink.h  function valid_channel()

return chan < MAVLINK_COMM_NUM_BUFFERS && mavlink_comm_port[chan]!=NULL;

#### Version
git

#### Platform
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover

night-ghost, 2017-02-18:
... and remove call to valid_channel in GCS_MAVLINK::init()

peterbarker, 2017-02-20:
@night-ghost Can this be replicated on master?

What is the problem you're trying to point out here?

I'm not trying to understand the disassembly just yet, but from your fix it seems you're of the understanding that mavlink_comm_port[chan] is null.

There is protection in the top of `Copter::mavlink_delay_cb` for port 0 - that check seems a touch silly to me.

More importantly, there are checks in `gcs_check_input` itself to ensure the channel is initialised.  You should not get to where you are in the code if `initialised` is not true, and that flag is set in `init`.

night-ghost, 2017-02-20:
initialization of mavlink_com_port at breakpoint still not done so access to mavlink_com_port[chan]-available() causes HardFault, and I don't know why all mentioned protections don't helps. it happens rarely, but it happens, and I spent a lot of time to understand why - so I think that to check in valid_channel() is not very bad thing. 

but thanks to this bug now I get in my HAL the address and code of exception by 3DR radio :)

[image]

night-ghost, 2017-02-20:
PS. may be it will be enough to move call to register_delay_callback() in system.cpp just before line

cliSerial->print("\nReady to FLY ");

night-ghost, 2017-02-20:
> Can this be replicated on master?

My HAL don't uses NUTTX so has a very different timing, moreover Master branch don't works on my hardware :)

peterbarker, 2017-02-20:
I really think we need to understand why those checks aren't working at the moment.  There are many places where we rely on initialised() to reflect the state of the object.

I'm concerned that if we make this change we may be hiding something very serious (memory overwrites of some description spring to mind).

Have you changed the threads-of-execution around as part of your FreeRTOS port?  Are there more threads involved here than on e.g. AP_HAL_PX4?

peterbarker, 2017-02-20:
BTW - love the exception over radio :-)

night-ghost, 2017-02-20:
>I really think we need to understand why those checks aren't working at the moment.

I try hard not to go into the code excepting the HAL, so the problems that come from there try to fix in the easiest way. So I did not investigate the the other problems, and simply removed the exception.

>part of your FreeRTOS port

I seem to said inaccurate. I do not use the OS at all. No NUTTX nor RTOS nor other. Bare metal. 
Return to Arduino times when all a bit familiar with the С language could easily change the code and compile own binaries. Without drivers, the command-line and other debris which comes with PixHawk.

So - no threads :) Timer interrupt as high priority and simple cooperative multitask for low priority needs - as it was on AtMega.

night-ghost, 2017-02-20:
this can be caused by calling GCS_MAVLINK::init(NULL) because library don't check the validity of the  port on input and set initialised in any case

magicrub, 2017-02-20:
Do you have a branch we can look at?

night-ghost, 2017-02-20:
You must be joking! On [Github](https://github.com/night-ghost/ardupilot), sure 

[discussion](https://www.rcgroups.com/forums/showthread.php?2784349-Ardupilot-on-OpenPilot-Revolution)

OXINARF, 2019-06-11:
@peterbarker Can we close this? AFAIK we don't have this issue on any of our supported HALs and we didn't really get to the bottom of why this was happening - which is unlikely to happen in a two and half year old issue.

closed 2019-06-11 by peterbarker