ArduPilot/ardupilot #5746: Problem with Gazebo simulation and ROS

BossKwei, 2017-02-17:
#### Platform
[  ] Copter in Gazebo (`sim_vehicle.py -f gazebo-iris -D --console --map`)
[  ] Gazebo 8
[  ] ROS Kinetic with Ubuntu 16.04

#### Issue details
Hi, I want to control an iris quadrotor with ros-mavros-package, I tried many times but failed in the end, so I really need some help.
There is a step-by-step guide of how I set it up:
1. `sim_vehicle.py -f gazebo-iris -D --console --map`
> Then console outputs: `SIM_VEHICLE: "mavproxy.py" "--master" "tcp:127.0.0.1:5760" "--sitl" "127.0.0.1:5501" "--out" "127.0.0.1:14550" "--out" "127.0.0.1:14551"`.
2. `gazebo --verbose worlds/iris_arducopter_demo.world`
> Now everything is ok
3. `arm throttle` & `rc 3 1700`
> Then the quadrotor began to takeoff
4. `roslaunch my_launch_file.launch`
> include file="$(find mavros)/launch/node.launch
> arg name="fcu_url" default="tcp://127.0.0.1:5760" 
> arg name="gcs_url" default="udp://127.0.0.1@127.0.0.1:14550" 
> arg name="tgt_system" default="1"
> arg name="tgt_component" default="1"
> arg name="log_output" default="screen"
> arg name="pluginlists_yaml" value="$(find mavros)/launch/apm_pluginlists.yaml"
> arg name="config_yaml" value="$(find mavros)/launch/apm_config.yaml"
5. mavros/mavlink could connect to sitl correctly, but when I tried to print some response such as `/diagnostics` topic, it always returns `fcu not connected`

Is there anything wrong in my step? I think I could not notice them by myself, I really need some help, thx

khancyr, 2017-02-17:
Hi, you mavros config is wrong :
arg name="fcu_url" default="tcp://127.0.0.1:5761" as 5760 is already take by mavproxy
you don't need arg name="gcs_url" default="udp://127.0.0.1@127.0.0.1:14550" as it is just for take out mavlink to a GCS but mavproxy is already doing in with "--out" "127.0.0.1:14550" "--out" "127.0.0.1:14551"

BTW :this issues list is for confirmed bugs and feature requests. User support questions should be posted at the forum (http://discuss.ardupilot.org).

OXINARF, 2017-02-17:
Like @khancyr said, this isn't the appropriate place for this. Either post in the forum or ask in Gitter.

closed 2017-02-17 by OXINARF

BossKwei, 2017-02-18:
@OXINARF @khancyr Thanks a lot.