ArduPilot/ardupilot #5749: AP_GPS: Log delta time between updates

WickedShell, 2017-02-17:
#### Issue details
When reviewing a log with high rate logging because of data loss its hard to distinguish if there was a loss of communication to the GPS or if it just lost some intermediate data. Tracking the delta between updates would allow us to state which happened with confidence

#### Version
Master

#### Platform
[ x ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover

rmackay9, 2017-06-08:
In addition, for Copter we should log an ERR message if it's slow.  Doing this makes it really obvious to someone analysing the logs at least if they're using Mission Planner which displays the errors very obviously.

gmorph, 2017-06-13:
why wouldn't we log a message for all vehicle types and not just copter?

WickedShell, 2017-09-15:
Done in #6556

closed 2017-09-15 by WickedShell