ArduPilot/ardupilot #5754: My plane crashs with nothing warning and abnormal?

ranqingfa, 2017-02-18:
It flys all well, and suddenly the motor stops, and with firmware 3.7 it seems a lot new strange problem happens.

3.7.1

[  ] All
[  ] AntennaTracker
[  ] Copter
[ x] Plane
[  ] Rover

fix wing

pixhawk

#### Logs
http://pan.baidu.com/s/1dFCBZQ5
TLOGS
http://pan.baidu.com/s/1pLpXbwn

ranqingfa, 2017-02-18:
crash video
http://pan.baidu.com/s/1eSsXkdk

ranqingfa, 2017-02-19:
same plane #5753

magicrub, 2017-02-19:
Please post help questions in the forum at discuss.ardupilot.org
The GitHub issues are for feature requests and confirmed/defined bugs.

closed 2017-02-19 by magicrub

ranqingfa, 2017-02-19:
This is not a bug?

magicrub, 2017-02-19:
If it is, what is it? This is not the area for troubleshooting. The forums are.

magicrub, 2017-02-19:
This log has 2 problems:
1) it doesn't have much in it so please enable your log bitmask so it will including helpful data
2) you're flying in MANUAL the whole time - the crash is pilot error.

ranqingfa, 2017-02-20:
But in tlog, it is in AUTO mode and crash in auto mode

ranqingfa, 2017-02-20:
Actually, its AUTO - FBWA - RTL - AUTO (switch waypoint) - crash, and Im confused too why only manual mode in log