ArduPilot/ardupilot #5762: Unable to get RNGFND parameters

bilarslan, 2017-02-21:
#### Issue details
Hi there,
I have an APM2.6 with lastest arduplane firmware and also Mission Planner. When i want to configure my rangefinder sonar sensor that Maxbotic Analog Sensor, there is no RNGFND params in full parameter list. There is only RNGFND_LANDING parameter. But when i install arducopter or rover firmwares, i get all of the RNGFND params and my sensor works perfect. How can i get RNGFND parameters when apm2.6 is installed arduplane firmware.

Thanks.

#### Version
Arduplane 3.7.1

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[ X ] Plane
[  ] Rover

#### Hardware type
APM 2.6

magicrub, 2017-02-21:
Hi @bilarslan The old 8bit APM 2.6 hardware has not had a firmware update in >2 years because. You are not running 3.7.1 on that hardware. Can you double check hardware/firmware please?

Chances are you're running old firmware and that old code does not have all the rangefinder features you're looking for. I suggest you upgrade your hardware so you can utilize the latest developments.

bilarslan, 2017-02-21:
@magicrub Thank you for your response.

magicrub, 2017-02-21:
if you do have APM 2.6 hardware, I'm sorry to inform you that it will not have any updates :( It's time to upgrade to modern hardware.

bilarslan, 2017-02-21:
Ok, it is time to buy an new hardware :)
Thank you so much @magicrub

closed 2017-02-21 by magicrub